cedricp/ddt4all #212: VIN

thomson56, 2019-06-02:
Hello, how to restore the number VIN after Virgin mode reset

shrlnm, 2019-06-02:
RadNav_*.xml
menu "Diagnosis/Identification" in the bottom of screen

thomson56, 2019-06-02:
Mais dans quel programme

Wysłane z aplikacji Poczta dla Windows 10

Od: shrlnm
Wysłano: niedziela, 2 czerwca 2019 20:02
Do: cedricp/ddt4all
DW: thomson56; Author
Temat: Re: [cedricp/ddt4all] VIN (#212)

RadNav_*.xml
menu "Diagnosis/Identification" in the bottom of screen

shrlnm, 2019-06-02:
> Mais dans quel programme

ddt4all or pyren/mod_ddt

thomson56, 2019-06-02:
Et vous pouvez utiliser DDT avec plus de précision, mais je ne sais pas où regarder, je ne le vois pas

Wysłane z aplikacji Poczta dla Windows 10

Od: shrlnm
Wysłano: niedziela, 2 czerwca 2019 20:48
Do: cedricp/ddt4all
DW: thomson56; Author
Temat: Re: [cedricp/ddt4all] VIN (#212)

Mais dans quel programme
ddt4all or pyren/mod_ddt

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif