cedricp/ddt4all #213: BCM/UCH FAl MMI Configuration

AndyIreland, 2019-06-02:
hello 
If I start up 

DDT4ALL in **DEMO mode** 'T4_VS_BCM_BIS_DDT2000_SW15_1'   'Fal_ MMI_Configuration'  
there is a setting for ECO_SWITCH_TYPE and parameters of 'NO BUTTON SWITCH'  'ECO BUTTON SWITCH' and 'MEX BUTTON SWITCH' 

But if I start T4_VS_BCM_BIS_DDT2000_SW15_1  Fal_ MMI_Configuration 
actually connect it on my car the ECO_SWITCH_TYPE does not show at all

Why is this please?

[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

monalos, 2025-10-03:
> [#836](https://github.com/cedricp/ddt4all/issues/836)

Can you delete a post entirely?