cedricp/ddt4all #215: Loss of touch response after applying mex

SnibboRen, 2019-06-04:
I have a Megane 4 with original rlink 2.2 upgraded to 3.3. I applied mex active and then performed a restart. Restart was quite slow and when finally up, the system no longer responded to touch inputs on the screen. All other functions are working correctly. Does anyone have any thought on  how to resolve?

pcboys, 2019-06-06:
I have encountered a similar situation, and then reset the default value in radnav /Engineering configuration, and restore it after reboot.
Perhaps you can find the same type of rlink2, refer to its default parameters.

SnibboRen, 2019-06-06:
Thanks @pcboys. When you say reset default,  are you referring to a particular variable or global defaults?

pcboys, 2019-06-09:
My vehicle is kadjar and the parameter values are not for your vehicle.

Dema323, 2020-06-02:
is mex = multisense ?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif