cedricp/ddt4all #216: The latest versions of the program for Android do not work.

duh555, 2019-06-06:
Hey. Version of the program for Android, starting with version 1.1 and ending with version 1.6 work fine. Versions 1.7, 1.71 and 1.72 start and immediately collapse. What is the problem?

cedricp, 2019-06-07:
No idea, could you plug your device to a PC and use Android tool 'logcat' to inspect the crash ?

atumelektronik, 2019-06-10:
[image]

duh555, 2019-06-10:
Hey. You probably do not understand me. I'm trying to use google translator. Version of my Android 5.1.1. Versions of Ecu Tweaker for Android, starting with version 1.1 and ending with version 1.6, work fine (1jpg). Versions 1.7, 1.71 and 1.72 are launched and immediately minimized (2jpg, 3jpg).
[image]
[image]
[image]

cedricp, 2019-06-10:
V1.73 released today, test it and report

duh555, 2019-06-10:
Скачал, установил - проблема осталась.

atumelektronik, 2019-06-16:
> Hey. Sürüm 1.1 ile başlayan ve sürüm 1.6 ile biten Android için programın sürümü iyi çalışıyor. 1.7, 1.71 ve 1.72 sürümleri başlar ve hemen daraltılır. Sorun nedir?

sorun bluetooth yazar tam olarak bluetooth'u adapte edemedi programa wifi kullananlarda sorun yok

duh555, 2019-06-16:
İpucu için teşekkürler. Wi-Fi adaptörü ile programın 1.73 sürümünü kontrol etti-program çalışıyor.

duh555, 2019-06-16:
Thanks for the advice "atumelektronik". He told me to use a Wi-Fi adapter. I tried the version of the program 1.73 with Wi-Fi adapter - it works well.

curiosopt, 2019-06-16:
The same happen on my OnePlus6, 1.7 and the latest versions of ecu tweaker stops like OP. Android 9

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif