cedricp/ddt4all #217: DDT4All - some technic question

bartekzsa, 2019-06-09:
First - Thank You for you work! is realy great!

I have question:
1. What is a proper and official name of this app? it's: ddt4all, DDT4All, DDT4all or maybe other name?
2. Where i can see application version when i run program?
3. Do you can share url for prefered DDT200 ecu database?
4. How to change color of app from black to white?

Regards.

cedricp, 2019-06-09:
1. DDT4All but others are Ok too
2. Check [release](https://github.com/cedricp/ddt4all/releases) page, but not sure to understand the question
3. No the database is copyrighted, you must own it
4. In last version you'll see a toggle button to switch to light theme

atumelektronik, 2019-06-10:
[[image]](url)

atumelektronik, 2019-06-10:
[Resim](https://i.hizliresim.com/LvYGoJ.jpg)

bartekzsa, 2019-06-15:
Thank you for answer.

2. I menad when i install DDT4All is any option to check in running program what version is it?
5. I don't understund why in this page https://github.com/cedricp/ddt4all/releases last version is 1.5 but in this website https://www.drive2.ru/b/498093336985338243/ last is 5.5.0 (i know that this version has integrated ddt2000 database) but question is on main software 1.5 vs 5.5?
6. Witch interface is THE BEST for DDT4All? Money don't care

Regards.

KarelSvo, 2019-06-21:
5.5.0means 550th change for improvement
550 commits  https://github.com/cedricp/ddt4all
https://github.com/cedricp/ddt4all/commits/master

https://www.amazon.com/ScanTool-OBDLink-USB-Professional-Diagnostics/dp/B005ZWM0R4/

mechatronicengineer, 2019-07-05:
I can't do active HSA option for my Dacia Sandero Stepway 2016 on Android App. I was able to change T_Pulse_CF and LINK_WITH_REVERSE_CF opsions but I get an error message when I change the HSA option

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif