cedricp/ddt4all #218: ELM Test not working

vicssan, 2019-06-11:
Hello, first of all, thank you for your great work. 
I have a problem connecting the elm. I have a Renault clio IV and when y test the ELM I have the folling text in the console window (I have made the cable modification  (ELM <=> CAR ...4-4; 5-5; 16-16; 6-13; 14-12)):
```
Usando base de datos DDT personalizada
libpng warning: iCCP: known incorrect sRGB profile
Intentando abrir puertoCOM3 @ 38400
Traceback (most recent call last):
  File "C:\Program Files (x86)\ddt4all\ddt4all.py", line 1159, in check_elm
    res = elm.elm_checker(port, speed, self.logview, core.QCoreApplication)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 1333, in elm_checker
    print cm[2] + " " + res.strip()
  File "C:\Program Files (x86)\ddt4all\Python27\lib\codecs.py", line 369, in write
    data, consumed = self.encode(object, self.errors)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range(128)
```

The ELM cable, without the modification, works ok with the ScanMaster-ELM.

Could you help me? 
I don't know if the elm has the PIC18F25K80 Chip. Could it be the reason?

Thank you!

cedricp, 2019-06-11:
If you're using the Windows installer, update the program by downloading the last source code, this bug should be fixed.
I don't think the ELM is the problem.

vicssan, 2019-06-11:
Hello again, I have updated with the source code but the error is still the same but the line numbers:
```
Usando base de datos DDT personalizada
Intentando abrir puertoCOM4 @ 38400
Traceback (most recent call last):
  File "C:\Program Files (x86)\ddt4all\ddt4all.py", line 1212, in check_elm
    res = elm.elm_checker(port, speed, self.logview, core.QCoreApplication)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 1389, in elm_checker
    print cm[2] + " " + res.strip()
  File "C:\Program Files (x86)\ddt4all\Python27\lib\codecs.py", line 369, in write
    data, consumed = self.encode(object, self.errors)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range(128)
```

In the log there are only these lines:

```
>[20:47:30.187]ATZ
<[0.809]ATZ
ü

ELM327 v1.5

>
>[20:47:30.996] ATZ                   
<[0.813] ATZ                   
ü

ELM327 v1.5

>
```

Thank you again for your help.

KarelSvo, 2019-06-15:
> ü

your elm is the problem

vicssan, 2019-07-18:
KarelSvo, you are right. I have tested it with the one you said and it works correctly. 

Thank you!

closed 2019-07-18 by vicssan