cedricp/ddt4all #221: ecu tweaker bluetooth the proplem

atumelektronik, 2019-06-15:
The author of the program could not solve the bluetooth problem. There is bluetooth problem in versions after 1.7. ecu tweaker stopped error. Turn off bluetooth is doing error when bluetooth is on.
[image]

cedricp, 2019-06-15:
I did some upgrades recently, try it

atumelektronik, 2019-06-15:
> Son zamanlarda bazı güncellemeler yaptım, dene

I've tried
[image]

tomidix, 2019-06-16:
@cedricp just close the issue and mention this BT problem as unsupported if this works on newer Android... Don't spend your time bothering with Android which was released October 31, 2013 when the cars are mostly newer...

cedricp, 2019-06-16:
This is probably not a big problem, but as I don't have an old Android device, I have no way to test it.
Could you try to connect your device to a PC and report logcat at the moment of the crash ?
https://developer.android.com/studio/command-line/logcat

cedricp, 2019-06-17:
I fixed this issue on version 1.74c.

atumelektronik, 2019-06-17:
> Bu sorunu 1.74c sürümünde düzelttim.

Dear cedricp DOWNLOAD LINK V1.74c has fixed the issue with this update. You are super. The correct diagnosis was easy to correct. I detected the Bluetooth problem and wrote it to you.

closed 2019-06-24 by cedricp