cedricp/ddt4all #222: Megane II ECU Reset Help me !!

nersoy72, 2019-06-16:
hello firstly. Yesterday with the error code in my tool ddt4all to delete the error connected to connect. However, after a short time, the fault came back (ELECTRONIC PROBLEM) and the SERVICE light was on. I accidentally disabled the tool while tampering with the program. Can't I reset ECU or delete ECU software? or I don't know if I may have deleted existing cards. Waiting for emergency help (backup card doesn't work either)

cedricp, 2019-06-17:
You'll need to be more precise : 
* Which ECU are you working on ?
* What's the ECU causing trouble ?
* What have you done ?
You cannot reset cards except if you virginize your BCM (UCH) ECU.

nersoy72, 2019-06-17:
At the top of the program Plugins> UCH Tools> Megane 2 UCH Reset command was applied. He's not reading both cards right now.

cedricp, 2019-06-17:
This tool is made to reset all UCH data (you got a virgin UCH now). You need to make an UCH learing from your card.
The problem is that it's now impossible to get APV code by asking UCH ISK code... 
It's clearly indicated in the plugin : **THIS PLUGIN WILL ERASE YOUR UCH**
If you tried to learn a card, you may try to parse your log file and search for frame 21AB.

nersoy72, 2019-06-18:
How should I proceed? Is there any possibility that you have copied the deleted software? If so, where can I find it? What is the 61AB framework you are talking about? Could you be more specific?

cedricp, 2019-06-18:
As I understand, you reset your UCH without knowning why you did it. The **reset UCH** function is a tool to **virginize** your UCH (when you get one from an old car) so you can use it in yours. It deletes all its configuration (cards, vehicle information, etc.).
The problem here is that you need **ISK** code to generate **APV** (PIN) code. **ISK** code is inside the **UCH** you can get it by requesting frame 21AB.
You need APV code to associate your cards to the car, but as your UCH as been cleared, I fear you can't do anything now.
If you have tried to do a card programming session before the reset, there's a chance to get **ISK** code from DDT4All's log file (it's in the application directory). Send me this file, I'll check if it's in-there.
What's the year of your car ?
I guess, from your understanding that it's going to be painful to get back to normal situation.

nersoy72, 2019-06-18:
Can I get the e-mail address to send the log file, please.

cedricp, 2019-06-18:
cedricpaille(at)gmail.com

nersoy72, 2019-06-18:
I just sent it.

nersoy72, 2019-06-18:
yes my car 2005

nersoy72, 2019-06-18:
Thank you. I'il try it now. I share the results.

nersoy72, 2019-06-19:
Hello Is it possible to identify and identify tire IDs through the program?

cedricp, 2019-06-19:
**UCH** -> **SSPP Configuration** -> **SSPP Apprentissage APV/retouche**
ARD -> rear right
ARL -> rear left
AVD -> front right
AVG -> front left
Did you manage to reprogram your cards ?

nersoy72, 2019-06-19:
Where can I find the ID of the tires?

Android için Outlook<https://aka.ms/ghei36>'u edinin

nersoy72, 2019-06-19:
The vehicle's UCH brain has been renewed. New card defined. The new UCH brain does not see the tires of the vehicle. How can I find out the ID numbers of the car's tire sensors?

nersoy72, 2019-06-19:
The vehicle's UCH brain has been renewed. New card defined. The new UCH brain does not see the tires of the vehicle. How can I find out the ID numbers of the car's tire sensors?
________________________________
Gönderen: Necati ERSOY
Gönderildi: 19 Haziran 2019 Çarşamba 16:49
Kime: cedricp/ddt4all
Konu: Re: [cedricp/ddt4all] Megane II ECU Reset Help me !! (#222)

Where can I find the ID of the tires?

Android için Outlook<https://aka.ms/ghei36>'u edinin

cedricp, 2019-06-19:
Two options :
1 - remove the tire and read the code on the TPMS device
2 - Activate the device with a special tool and read the value on the SSPP screen with learning button

nersoy72, 2019-06-20:
Hi, I'm sending e-mail a picture of the tire pressure sensor. Can you tell me which part is the tire ID number ?

nersoy72, 2019-06-20:
I've recorded the tire pressure sensors in the UCH computer. The problem has improved. Thank you so much :)

cedricp, 2019-06-24:
Cool, so I close the ticket

closed 2019-06-24 by cedricp

Nikola87, 2020-10-06:
hello
please help
yesterday i do the same with UHC on my renault scenic 2 2005.
Now when I want to make a new key don't give me a ISK and Pin.
I use Renolink 1.52 and now i download DDT4all

Dawid198881, 2024-06-30:
I have the same issue. Can you help me?