cedricp/ddt4all #224: I want to hear the horn beep when I lock the doors.

atumelektronik, 2019-06-18:
renault megane 3 I want to hear the horn beep when I lock the doors. Can I do this with ddt4all?

atumelektronik, 2019-06-18:
sonunda ddtall ile arabamın kapısını kilitleyince korna bip yapıyor ayarladım.

tanerdogan, 2019-06-19:
Çözümü yazarsanız biz de deneyebiliriz?

pcboys, 2019-07-09:
bcm/uch ->AlarmPerimetric->AlarmPerimetric-configuration ->BIPBIP_LOCK_CF ->true

AndyIreland, 2019-07-09:
I have the Perimetric alarm enabled and it flashes the indicators for a while if someone where to break in , but no horn sounds - does anyone know why?

EXcomZkko, 2019-07-19:
@atumelektronik  Hocam nasıl yaptınız, biraz detay verebilir misiniz? Önceden beri istediğim bir olay.

BarektE, 2019-09-23:
bu sayfaya yazanlar bilgi sahibi değiller. sayfa yararlı değik

closed 2019-09-23 by BarektE