cedricp/ddt4all #226: PDC (UPA Valeo 2.0) ecu reading datas

hardweb-it, 2019-06-24:
Hi all, I'm trying to read the Parking Distance control's values directly from canbus broadcast, using an Arduino nano and an mcp2515 unit.
Actually i found the AT commands between ddt4all and ELM237 interface, but I would find the PID of the PDC communication in order to get the values.

I can't understand all parts of communication, btw that's an example:

74E;222037;62 20 37 62 FF FF B7 AD AE B9 9C 9C 80

I think it's something like this:
0xPID (right?)
222037 (I don't know what this is)
HEX of calculated distances where FF = 250 centimeters, to 00 = 0 centimeters.

Someone can help me?
Thanks in advance

cedricp, 2019-06-24:
At first, you must use ISOTP library (https://github.com/altelch/iso-tp or another one) for your arduino to communicate in diagnostic mode. 
Then, study the requests with DDT4All in edition mode and "request" and data tabs. You'll find how frames are calculated from/to data.
If you want to work with broadcast messages, check the "can sniffer" tab.
I think you have to study CAN bus a little bit more ;)

cedricp, 2019-06-24:
Here's an exemple : 

[image]

You get the complete details of how the frame is built. Right click on the data name and "goto data" to get a full description of the data conversion.

cedricp, 2019-06-24:
74E;222037;62 20 37 62 FF FF B7 AD AE B9 9C 9C 80
74E -> address of the ECU
22 20 37 -> PID
62 20 37 62 FF FF B7 AD AE B9 9C 9C 80 -> ECU response to frame above, 62 20 means positive response (0x22 + 0x40 = 0x62), the rest is the data of that PID request

hardweb-it, 2019-06-25:
THANK YOU so much Cedric! You are the best 🥇 
I just started to understand the CAN communication, and I've a lot to study but with your help now I know what I need to do, and I understood how to read the logs. When my project is complete I will share it with all. Thanks alot!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif