cedricp/ddt4all #228: OVERLOAD_BY_SMART_CF

kazaa8, 2019-06-30:
Hello,
when I try to replace LED bulbs in turn signal, controller detects infinite resistance and fast flashers turn on. I try to turn off fast flashers and I found parameter: 
Flashers > Flashers - Configuration -> OVERLOAD_BY_SMART_CF

Anybody know how it works? I tried to turn off this (default is 1) but nothing change in front led bulbs.

SnibboRen, 2019-07-03:
If I remember correctly, the Italian Facebook site Megane Passion has a description of this. megane4-forum.de would also be a place to look.

sirbig, 2020-10-15:
Hello @kazaa8, I'm also interested to know if there is a way to cancel this canbus error, did you find a way to solve this issue?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif