cedricp/ddt4all #230: Mirror autofolding Kaptur rus 2019

Gres1978, 2019-07-04:
I have Kaptur rus 2019
Through the program ddt4all disable mirror autofolding - UPC-EMM / EMM_EDISON_DDT2000_SW15_2_V1.0 / EMM_AutoFolding / EMM_AutoFolding-Confiuration = Nbx_AutoFolding_CF = 0

But when you turn off the auto-folding function of the mirrors, it turns out that manually when you press a button in the salon, the mirrors are added, but they are not folded back. It is necessary to re-enable auto-folding.
How to provide manual folding mirrors with a button from the salon?

AndyIreland, 2019-07-10:
are they not supposed to autofold automatically when you turn the engine off and get out of the car or press the lock key on the keycard? - have you tried pressing lock button on keycard twice or long press  and see if that works?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif