cedricp/ddt4all #233: "Please add EOL settings" after Configuration Reset - PLEASE HELP :'(

Royan85, 2019-07-17:
Hi, 
I have enabled Android Auto in several vehicles using ddt4all from a while, but today one failed :(. I saved the Android Auto and Mirrorlink to "Present" value, but when I did Reset Hard, the nav entered in a bootloop only showing the TomTom logo.
So I tried to reset the entire ECU (MFD v4.6) to default by pressing "Reset Configuration" button, but now the system is stuck at a warning screen saying "Unconfigured device, please add EOL settings".
I am somewhat desperated because it is not my car, and if the local Renault dealer find a way to fix it, I bet they will charge us a fortune.
Please if you know a way to fixit via the ddt4all software, I would be very very very thankful.

Thanks in advance,
Raul

atumelektronik, 2019-07-17:
> Merhaba, 
> ddt4all'u bir süredir kullanan birçok araçta Android Auto'yu etkinleştirdim, ancak bugün başarısız oldu :(. Android Auto ve Mirrorlink'i "Present" değerine kaydettim, ancak Sert Sıfırla yaptığımda, nav yalnızca bir önyükleme işlemine girdi TomTom logosunu göstererek, 
> "Yapılandırmayı Sıfırla" düğmesine basarak tüm ECU'yu (MFD v4.6) varsayılana sıfırlamaya çalıştım, ancak şimdi sistem "Yapılandırılmamış cihaz, lütfen EOL ayarları ekle" diyen bir uyarı ekranında kaldı. 
> Biraz zorlandım, çünkü bu benim arabam değil ve yerel Renault bayii tamir etmenin bir yolunu bulursa, bize bir servet talep edeceklerdir, 
> lütfen ddt4all yazılımı ile düzeltmenin bir yolunu biliyorsanız lütfen çok çok müteşekkirim.
> 
> Şimdiden teşekkürler, 
> Raul

ddt4all ile düzeltmeniz imkansız. Serviste yazılım atılacak başka çaresi yok.

Royan85, 2019-07-18:
> ddt4all ile düzeltmeniz imkansız. Serviste yazılım atılacak başka çaresi yok.

Thanks for your response. We are going to the local dealer to try for a ECU reprogramation. Hope we can solve this.

Royan85, 2019-07-18:
Fortunately, the dealer successfully managed to reprogram de navigation ECU and everything is working fine again.

atumelektronik, 2019-07-19:
> Neyse ki, satıcı başarıyla navigasyon ECU'sunu yeniden programlamayı başardı ve her şey tekrar iyi çalışıyor.

DDT4all ile kesinlikle radyo ve navigasyon ECU'suna girmeyin anında sıfırlıyor rayoyu. Sonra yetkili serviste yazılımı yeniden yükletmek zorunda kalırsınız

ceboline, 2019-07-19:
Atumelekronik merhaba
Bilgilendirme için teşekkürler
Radyo ve navigasyon ECU’ nun isimleri ne ?
Girmememiz gereken ECU ları yazarmısım
Teşekkürler

Saygılarımla/Mit freundlichen Grüssen/ Best Regards.

Cebrail GŪZEL

Quality  MB Trucks Turkey
 ( TE/QMT )
Foreman of the Vehicle final checks & Delivery ınspection

Mercedes-Benz Türk A.Ş
Aratol Bahçeli mahallesi 157/Ahmet Davutoğlu bulvarı no:11
68100 Aksaray / Türkiye
0506 604 38 72
[X]

atumelektronik, 2019-07-19:
Bunun gibi daha birçok yer teybin sıfırlanmasını sağlıyor 13 defa teyp sıfırladım. tecrübe konuşuyor
[image]
[image]
[image]

nazmi02551, 2019-07-19:
> > Neyse ki, satıcı başarıyla navigasyon ECU'sunu yeniden programlamayı başardı ve her şey tekrar iyi çalışıyor.
> 
> DDT4all ile kesinlikle radyo ve navigasyon ECU'suna girmeyin anında sıfırlıyor rayoyu. Sonra yetkili serviste yazılımı yeniden yükletmek zorunda kalırsınız
Yanlış bir işlem yapılmazsa sadece ECU'ya girmekle sıfırlanma söz konusu değil. Daha doğrusu ben defalarca girmeme ve işlem de yapmama rağmen böyle bir şeyle karşılaşmadım.

Royan85, 2019-07-19:
Indeed, my mistake was to pressing the "Configuration Reset" button. I can set/unset "Android Auto" variable several times without wiping the ECU data.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif