cedricp/ddt4all #236: Modo virgen R link 2 megane 4

Shadowever, 2019-07-25:
Hola, necesito ayuda para salir del modo virgen.

shrlnm, 2019-07-25:
Share your email please

Shadowever, 2019-07-25:
Shadowever25@hotmail.com

Emaromao123, 2020-08-11:
Good Evening,

Same problem, ive pressed virgin mode so what now?
I have Usb Elm interface, already tried pyren with the same macros and nothing, it seems that i cant have acess to CAN.

Any help? 

Ema-pt@hotmail.com

Emaromao123, 2020-08-12:
> Share your email please

Can you help me getting out of virgin mode? Also tried pyren with the macros and nothing happens. I have wrong response with can. I'm using elm 327 USB. Can you help? Ema-pt@hotmail.com.
Thank you

Rubele2, 2020-12-23:
Hola!
Necesito ayuda para reestablecer la pantalla del Rlink . Le he dado a Virgin Mode sin querer, he probado con las 4 macros de pyren y no funciona. Podrías ayudarme por favor?
Mi correo es rubele@hotmail.es.
Muchas gracias

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif