cedricp/ddt4all #237: unknown Navigation on R-Link

flegaigor, 2019-08-01:
hi,
today I have tried to turn on rear camera on my Renault megane III ph2. I made a change to Can bus wires and scan my car computers. DDT4all found Audio(radio) and MIU and Navigation with no relevant ECU file found: Diagversion [20] Supplier [329] Soft [333C] Version [4130].
How can I access this navigation and make changes?

myshuker, 2021-04-04:
did you solved the problem ? and how if yes

AndyIreland, 2022-06-21:
> Hello, I have a r link after the reset does not want to run, black screen. Via the modified cable both DDT4all and Reno link cannot find computers from navigation or radio. Can somebody help me?

does screen light up even if its black or blank, can you see the backlight of the screen. and try pressing home button 5 times . and try disconnecting 12v battery under bonnet for 10 mins then connect back up (if it does start up you may need to enter the r-link security code)
and you are sure the rewired lead or adaptor is wired properly yep?

vraug666, 2022-06-21:
> 

I used to activate android car via modified cable. After that I found the upgrade, installed and the r link no longer wants to run. The car is a megane3 2015.
 Installed new radio, display, BIC, UCH and again does not respond. I don't know why I can't find computers from navi, radio and BIC.

AndyIreland, 2022-06-21:
and does backlight of screen light up?

vraug666, 2022-06-21:
> i svijetli li pozadinsko osvjetljenje ekrana?

Not. Everything is dead.

AndyIreland, 2022-06-21:
checked the rlink fuse in the fuse box?

vraug666, 2022-06-21:
Yes

AndyIreland, 2022-06-21:
try pyren and tick box 'can2' and try searching , if not then sounds like boot img of rlink corrupted and needs re-flashing on a bench

vraug666, 2022-06-21:
Ok. How to flash it again.

vraug666, 2022-06-21:
> try pyren and tick box 'can2' and try searching , if not then sounds like boot img of rlink corrupted and needs re-flashing on a bench 
With a modified cable I scan or not?

AndyIreland, 2022-06-21:
yep, use modified cable and select can2 box in pyren

vraug666, 2022-06-21:
> da, koristite modificirani kabel i odaberite can2 box u pyrenu

OK

vraug666, 2022-06-21:
> Ok. How to flash it again.

AndyIreland, 2022-06-21:
well i dunno how to do it but this Guy in Romania I think fixes them and re-flashes them if you drop him a line see what he says:
https://www.facebook.com/iongeorgel.popa

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif