cedricp/ddt4all #238: UCH NOT READY

tolbiac, 2019-08-02:
Bonjour, merci pour ce beau travail sur ddt4all. Mon objectif est de reprogrammer une clé pour un scénic 2 (2005) pour mon propre usage. J'ai donc trouvé ddt4all et je me suis équipé d'un elm.

La procédure n'est pas très évidente, j'ai trouvé une vidéo l'expliquant, elle est faite avec une version antérieur de ddt4all (la fonction ISK to PIN n'est pas visible dans la version que j'utilise par exemple).

Du coup je ne suis pas certain de ma manip et j'ai besoin d'un peu d'aide... 

Obtenir l'ISK et le PIN : par le menu Extensions->Clefs. Je n'obtiens que "UCH NOT READY"

De même dans la fonction APPRENTISSAGE APRES-VENTE 2.0 ->tester virginité du badge, les logs indiquent qu'il n'y a pas de réponse de l'ELM 
[image]

J'ai mis en doute l'ELM, mais en utilisant en version démo EOBD-F l'appli se connecte bien, je pense donc que l'adaptateur ELM fonctionne bien.

Y a t il un moyen de confirmer que la communication est bonne? qu'en pensez vous? merci d'avance pour vos retours

cedricp, 2019-08-03:
Je pense que c'est  un probleme d'ELM (clone). Certains n'ont pas les fonctionnalités avancées suffisantes pour se connecter aux ECUs. Je l'ai déjà constaté

tolbiac, 2019-08-03:
Merci pour ton rapide retour, je regarde ça

tolbiac, 2019-08-27:
Bonjour, après réception d'un nouvel ELM, la manipulation a réussi. Je suis pas certain de la procédure, les valeurs dans "Armer le service après vente" et Validation restaient à 0... La clé fonctionne c'est le principal. Encore merci pour l'outil et ton support!

closed 2019-08-27 by tolbiac

tolbiac, 2019-09-05:
Bonjour, ma clé d'origine qui fonctionnait encore ne fonctionne plus du tout, est ce un effet de bord habituel? merci d'avance pour ton retour

cedricp, 2019-09-05:
Il faut repasser toutes les clefs lors de la reprog, celles déjà
enregistrées sont perdues.

tolbiac, 2019-09-05:
ok merci pour l'info!