cedricp/ddt4all #240: The megane 2 card does not lock the doors.

nersoy72, 2019-08-05:
Hello there. The megane 2 card is running in my car but does not lock the doors. How can I fix ?

cedricp, 2019-08-20:
Is it a handsfree card ?

nersoy72, 2019-08-20:
starts the car but does not lock the doors

Windows 10 için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

________________________________
Gönderen: Cedric PAILLE <notifications@github.com>
Gönderildi: Tuesday, August 20, 2019 10:34:38 PM
Kime: cedricp/ddt4all <ddt4all@noreply.github.com>
Bilgi: nersoy72 <pilot1453@hotmail.com>; Author <author@noreply.github.com>
Konu: Re: [cedricp/ddt4all] The megane 2 card does not lock the doors. (#240)

Is it a handsfree card ?

Icereturns, 2019-10-23:
Almost certainly the battery in your key card is dead. It does not require power to start the car when placed in the slot.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif