cedricp/ddt4all #242: R-Link2 Update Errors

mavvi, 2019-08-15:
Hello. Please help...
Update v7 after
The camera is not active when reverse gear is engaged after the R-Link update.
The climate display does not show the set temperature.

atumelektronik, 2019-09-01:
EEPROM'da ilk kapalı itme dikiş şeridinin girdap Flap Bayrağı
EEPROM'da açılan ilk açılan itme dikiş şeridinin Flap Bayrağı
Swirl Flap İlk kapalı baskı pozisyonu leamt'ın mevcut değeri
Swirl Flap Son kapalı baskı pozisyonu 'leamt'ın akım değeri
Swirl Flap 'Açılmış ilk itme pozisyonu gergisinin güncel değeri
Swirl Flap Açık açılan son konumdaki 'leamt

[image]

comaly, 2019-09-01:
Merhabalar,
Bu konularda tecrubeli değilim.
Anlayabileceğim şekilde yazabilir misiniz?
Aracım Megane 4 HB ICON EDC DCI 2017 model

atumelektronik, 2019-09-02:
bunları kontrol edin

[image]
[image]

comaly, 2019-09-20:
Sorun halen devam ediyor. / The problem still exists.

yk2548, 2020-03-31:
merhaba sorunu çözebildiniz mi? aynı sorunu yaşıyorum şuan yardımcı olursanız sevinirim..

yk2548, 2020-04-06:
ben servise gittim aynı sorun vardı yarım saat içinde yeniden programlayıp verdiler 63tl ücret aldılar. ddt4all ile yapılacak birşey değil. yükseltme yaparken eksik kalan birşeyler var ön yükleyici sürümü. bu sürümün güncellenmesi gerekli ve bunu sadece yetkili servis yapabiliyor. v2 den v7 ye kendim yükseltmiştim.

BarektE, 2020-04-06:
> ben servise gittim aynı sorun vardı yarım saat içinde yeniden programlayıp verdiler 63tl ücret aldılar. ddt4all ile yapılacak birşey değil. yükseltme yaparken eksik kalan birşeyler var ön yükleyici [...]

Hangi servis olduğunu paylaşmanız mümkün mü? 63TL çok uygun

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif