cedricp/ddt4all #243: No remote AC or charge schedule after activate clock on dashboard

MarcosMar, 2019-08-16:
Hello.
I have a Zoe.
After do the mod to activate the clock and the outside temp in the dashboard, and choose the option "No mfd present" (because the clock is under the instant consummation), I really that I can't no more activate by the Renault app the remote AC, Ac schedule and charge schedule!
I put settings back but still don't work remotely, just on the car.

I saw that a lot of people that do this mode have the same issue..... Anyone have solved this problem ?

Sorry my English....

MarcosMar, 2019-08-16:
I know that's not a ddt4all issue but if anyone have solved this problem please let me know.
Thanks

KarelSvo, 2019-08-26:
No log file available?

MarcosMar, 2019-09-14:
Sorry for my late answer.
I cannot retrieve the log.
I made the mode with the ecutweker android app and when I press view log, the app freezes and after 2 min suddenly close!

cedricp, 2019-09-16:
You can get your logs in your Android device with adb or a file explorer here :
/data/data/org.quark.dr.canapp/files/log.txt
When the file is too big, it can overflow the clipboard.

MarcosMar, 2019-09-16:
I open the discribed folder but it shows 0 files! ...

cedricp, 2019-09-17:
Yes, you probably don't have permission to read it or you cleared it. I released v1.8, this version copies the log file to your download directory instead of showing it (and doesn't crash if log is too big !).

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif