cedricp/ddt4all #247: Connection issues

thodorisgo, 2019-09-06:
Hello and congrats for your great work.
I'm trying to connect on my Renault Megane IV. I use ELM327 v1.5 usb
[This one](https://m.aliexpress.com/item/32319918190.html)  
I tried to connect via Win10 and Win8 and at the beggining i get the error that ELM not responding and i can't connect. 
[image]

I used the ScanMaster software and the connection is working fine. After making the connection on ScanMaster i try again ddt4all and i managed to connect but when i ooen the cluster the right column is red and says NO DATA.
[image]
Can you please suggest me a solution?
Thank you very much

BarektE, 2019-09-06:
Device of the computer. manager. Remove OBD 2 there. Then the update will be corrected

thodorisgo, 2019-09-06:
Thank you very much buddy but i struggled with the drivers it was not easy. Do you know any working drivers for windows 10?

BarektE, 2019-09-06:
Kablolu OBD 2 bende benzer sorunu yaşıyorum. Herseferinde driver güncelliyorum

thodorisgo, 2019-09-06:
Kanka ben yunan i dont understand turkish

BarektE, 2019-09-07:
Google çeviri kullan

Χρησιμοποιήστε τη μετάφραση Google ·)

thodorisgo, 2019-09-07:
Guys after doing some research i noticed that the elm327 i own its not the original with the pic18f2480 chipset. Might this cause my problems ?

cedricp, 2019-09-07:
Absolutely

KarelSvo, 2019-09-07:
> Guys after doing some research i noticed that the elm327 i own its not the original with the pic18f2480 chipset. Might this cause my problems ?

When the FTDI chip is cloned.
Do not use the automatic driver installation of the Windows update function.
But disable the update automatic and use the FTDI driver from the Elm CD.

https://www.amazon.com/ScanTool-OBDLink-USB-Professional-Diagnostics/dp/B005ZWM0R4/
This USB interface works perfectly, since the original FTDI chip and original AT command set are available.
https://www.scantool.net/downloads/updates/obdlink_sx/
There you will find firmware support for this interface.

thodorisgo, 2019-09-07:
Every Elm327 thath i found in greece is Chinese replica so i ordered this one from Italy
https://www.amazon.it/iLC-Wireless-Lettore-Scansione-Strumento/dp/B01NAQ2878/ref=mp_s_a_1_2?keywords=elm+327&qid=1567847800&s=gateway&smid=ASYC0WZBA5NIH&sprefix=elm&sr=8-2

thodorisgo, 2019-09-19:
Hello again the odb that i mentioned on my past message arrived and works great. Just letting you know that is compatible.
Regards

closed 2019-09-19 by thodorisgo