cedricp/ddt4all #249: DDT4ALL - New full installer request

bovirus, 2019-09-09:
@cedricp 

Could you please update the full installer (with all files and ECU)?

the latest build is dated November 2018 (quite 1 year ago).

Could you please also make a multi-language installer (I can help about it)?

The file in this post

https://www.drive2.ru/b/498093336985338243/

are not available anymore.
Do you know as we can contact Russian author?

Thanks.

cedricp, 2019-09-09:
I'll try to generate a new installer with updates ASAP, but I cannot inclue DB files because they are copyrighted.
You wanna contact Zmiterm ?

bovirus, 2019-09-09:
@cedricp 

Yes. If it possible I want to contact Zmiterm.

AndyIreland, 2019-09-09:
> @cedricp
>
> Could you please update the full installer (with all files and ECU)?
>
> the latest build is dated November 2018 (quite 1 year ago).
>
> Could you please also make a multi-language installer (I [...]

mine is pretty up to date version of 5.5.0 and ECU DB version of  9/7/2019 

[https://cloud.mail.ru/public/4mtR/2PLQ7JcmG](url)

bovirus, 2019-09-10:
@AndyIreland 

The second link point to github.

How can contact the author (Zmiterm) of 

https://www.drive2.ru/b/498093336985338243/

I want to make the installer multilanguage..

KarelSvo, 2019-09-16:
> How can contact the author (Zmiterm)

In the profile click on the button Cообщение.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif