cedricp/ddt4all #250: For those having trouble getting Welcome/Goodbye lights to work on their Clio 4

AndyIreland, 2019-09-09:
hello - I just wanted to be helpful and relay the findings I have regarding activating Welcome and Goodbye Lights feature for those without LED lighting or Intens clio. 

My Clio Mk IV is a Dymanique 2016 model with halogen Head-Lights and LED DRL in the Radiator Grille and EMM of : EMM EDISON DDT2000 SW15_2 V1.0

Here are my findings:

The setting that seems to do the magic is :

NSX Singlebulb DRL TAIL CF - it has value of min. 0 and Max 3

If you set this to :

0 = no Driving lights (DRL)
1= DRL on with Auto (DRL go out when headlights are turned on)
2= DRL on Full Brightness (but DIM when Headlights are turned on) 
3= DRL on Full Brightness (and still on full brightness when headlights are on) But - Welcome & Goodbye lights Feature works (the DRL's Stay on for a while when you unlock the door or when you walk away from the car ) = Welcome & Goodbye lights feature 

Hope this helps people for when even they have enabled 'welcome & Goodbye lights on their non intens/non LED headlamps clio in DDT4ALL and still not get the welcome & Goodbye light function to work 

-------------------------------------------------------------------------

bonjour - je voulais juste vous aider et faire part de mes conclusions concernant l’activation de la fonction Welcome et Goodbye Lights pour les personnes n’ayant pas d’éclairage LED ni d’Intens clio.

My Clio Mk IV est un modèle Dymanique 2016 doté de phares à halogène et de diodes électroluminescentes (DRL) dans la grille de radiateur et d'EMM de: EMM EDISON DDT2000 SW15_2 V1.0

Voici mes conclusions:

Le paramètre qui semble faire la magie est:

NSX Singlebulb DRL TAIL CF - sa valeur est min. 0 et 3 max

Si vous définissez ceci sur:

0 = pas de feux de circulation (DRL)
1 = DRL activé avec Auto (DRL s'éteint lorsque les phares sont allumés)
2 = DRL en pleine luminosité (mais DIM lorsque les phares sont allumés)
3 = DRL en pleine luminosité (et toujours en pleine luminosité lorsque les phares sont allumés) Mais - bienvenue et au revoir lumières fonction fonctionne (les DRL restent allumés pendant un moment lorsque vous déverrouillez la porte ou lorsque vous vous éloignez de la voiture) = Welcome & Au revoir lumières

J'espère que cela aidera les gens quand même ils auront activé 'les lumières de bienvenue et d'adieu sur leurs projecteurs non intensifs / non à LED en DDT4ALL et que la fonction de bienvenue et d'adieu ne fonctionne toujours pas

[image]

EXcomZkko, 2019-09-24:
> 
> 
> Hi Andy,
> 
> This is a good find but I heard somewhere that this should only be attempted with SW15+ otherwise might cause really bad issues?

Yes. I'm not sure about more expensive devices (like obdlink) than elm327's. But you should not touch any parameters on "EMM_Lighting - Configuration" section lower than v15, if you don't have Renault Clip device near you.

AndyIreland, 2019-09-24:
ah yes indeed , I think its sw14 and above and hear that if you enable 'see me home' / 'follow me home' on under SW14 you can indeed kill all your lighting and ECU will need reprogramming 
I did put in original post that mine was EDISON DDT2000 SW15_2 V1.0 but maybe I should have explained it better or give warning of not to try under SW15.
Thanks.

ClioIV, 2020-05-14:
hello I have a big problem I use the SW15 and the headlights do not work! can you help me please

AndyIreland, 2020-05-14:
> hello I have a big problem I use the SW15 and the headlights do not work! can you help me please

Hello - if you have put all parameters as before and lights do not work try this:

You will need your radio code (sometimes written in owners manual in pen or pencil) because we are going to disconnect the battery. 

Disconnect the negative terminal on battery. Put headlamp switch from 'auto' to headlamp. Pull beam headlamp stalk leave car with door open for 10 minutes or more or overnight in garage if you can. We are hoping that will 'drain' all power from the lighting ECU and reset it . 

Put lights switch stalk back to 'auto' (or off) - attach the negative back onto the battery , put radio code back in to get radio working again. Start up engine and then check headlights and see if they work. 

let me know how you get on .

ClioIV, 2020-05-15:
J’ai fait un scan auto et je suis en version 10

ClioIV, 2020-05-15:
[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif