cedricp/ddt4all #253: Can we lock the doors till the car is running?

vafko, 2019-09-12:
Hello, can we make this with keyless card on Fluence to lock the doors, when the engine is working? It will be very useful in the hot summer days. I have other car (not Renault) that this is working just fine.

Thank you!

thodorisgo, 2019-09-19:
Rlink has built in option to lock the doors when you reach 20km. I think thats the only option. But what closing doors has to do with hot summer anyways ?

mnigten, 2019-09-19:
I think to put airco on 10 minutes before leaving 😉
Same wish here

AndyIreland, 2019-10-21:
can RES (remote engine start) be configured with the  Fluence and DDT4ALL? - then you could maybe put AC on and start up the engine remotely with keycard without opening doors?

kenanyildiz55, 2020-05-19:
@AndyIreland Yes, it can be operated remotely. Using DDT4all. However, I do not know how. I'm throwing a video link.

https://www.youtube.com/watch?v=LPq0YwMq8B0

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

MiqdadAli2002, 2024-06-14:
> @AndyIreland Yes, it can be operated remotely. Using DDT4all. However, I do not know how. I'm throwing a video link.
> 
> https://www.youtube.com/watch?v=LPq0YwMq8B0

This video is private
Could you help me with programming or another video