cedricp/ddt4all #255: Is the EGR shown to the car with Ddt4all?

BarektE, 2019-09-16:
**I wrote in 6 (six) languages. I hope someone knows**

**English**
After EGR purging, the injection (brain) should show EGR. Renault service shows (CAN CLIP). Is it possible to do this with DDT4all?

**Руска**
После продувки EGR впрыск (мозг) должен показывать EGR. Показывает сервис Renault (CAN CLIP). Возможно ли это сделать с помощью DDT4all?

**Español**
Después de la purga de EGR, la inyección (cerebro) debe mostrar EGR. Servicio de Renault muestra (CAN CLIP). ¿Es posible hacer esto con DDT4all?

**İtaliano**
Dopo l'eliminazione dell'EGR, l'iniezione (cervello) dovrebbe mostrare l'EGR. Manifestazioni di servizio Renault (CAN CLIP). È possibile farlo con DDT4all?

**Français**
Après la purge de l'EGR, l'injection (cerveau) doit montrer EGR. Service shows Renault (CAN CLIP). Est-il possible de faire cela avec DDT4all?

**Türkçe**
EGR temizlemesinden sonra, EGR arabanın beynine bura senin yerin diye gösterilmesi gerekiyormuş. Renault servisi söyledi. Onlar (CAN CLIP)ile gösteriyorlarmış. Bunu DDT4all ile yapmak mümkün mü?

shrlnm, 2019-09-16:
If CLIP may do it then ddt4all may do it too for sure.
For answering "how" we need three things
1) CLIP ECU id (5 digits in bottom of CLIP screen)
2) CLIP Parameter name (something like PR057)
3) ddt ecu name xml/json file name

BarektE, 2019-09-16:
Thank you for your answer. car 2011 megane 3 x38 With Ddt4all, which (ECU), will we enter, do the parameters?

BarektE, 2019-09-16:
Is there someone who uses ddt4all, who knows that?

BarektE, 2019-09-16:
bilgisi olan yok mu?

BarektE, 2019-09-23:
bu sayfaya yazanlar bilgi sahibi değiller. sayfa yararlı değik

closed 2019-09-23 by BarektE