cedricp/ddt4all #256: config generale 3 in TDB cluster clio 4

AndyIreland, 2019-09-16:
I have just looked in demo of DDT4ALL and see a config generale 3 section in TDB cluster . I had only seen config generale 1 & 2 before. its the first time I have seen this 3rd page. 

It says about 'water temp'  in the ADAC , does this mean if its enabled will the water temperature of the engine coolant wil be displayed on the satnav screen or scrolling through the info window on the speedo cluster? 

[https://www.pcrepairs.x10host.com/water_temp.jpg](url)

[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

AndyIreland, 2025-10-31:
> you managed to Activate coolant temperature on dash ?

no , it didnt do anything for me when i enabled it

AndyIreland, 2026-03-25:
> I can't see ddt4all cluster tdb to change from mpg to L in my Megane IV. Any helps

This , if this is your correct found ecu?:

[image]