cedricp/ddt4all #258: Megane IV - Update Error

comaly, 2019-09-20:
Please help me.
Can't update or downgrade R-link.
I get the error in the picture in r-link all versions.

[image]

My Car: Megane 4 HB EDC DCI 2017

shrlnm, 2019-09-20:
You'd better ask here https://www.msieurlolo.fr/forum/index.php?board=678.0

comaly, 2019-09-22:
> You'd better ask here https://www.msieurlolo.fr/forum/index.php?board=678.0

"Le sujet ou la section que vous recherchez à l'air d'être manquant ou inaccessible pour vous."

I'm a member but the page you specified doesn't open.

jaflo89, 2020-08-27:
> Please help me.
> Can't update or downgrade R-link.
> I get the error in the picture in r-link all versions.
> 
> [image]
> 
> My Car: Megane 4 HB EDC DCI 2017

Hello! How was it solved? THx

Tony2764, 2020-08-29:
Bonjour, 

Avez vous résolu le problème ?
Cordialement

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif