cedricp/ddt4all #259: Vehicle ID problem (RLink blocked)

igno77, 2019-09-21:
Hi to all,
I need a big help for my RLink 2! The situation is the follow:
I've received from a my friend that live far from me (500km) and work in Renault a completely new RLink2 v8.0 with DAB for replace my RLink2 v3.3 without DAB. For activate the new one he had putted the RLink on another Megane IV, connected with canclip and finally wrote my VIN code in order to be placed on my car. On that Megane it work fine, it's fantastic!! BUT, yesterday I had received the new RLink and when I had putted on my car I receive the message "System Blocked for Antithief function"! It's very weird because inside there is my VIN code (cheked with ddt4all) but, after a lot of research I think that I have find the probem. There is another code called Vehicle ID present in RadNav and it must match with the same number in the UCH ecu. Of course it was the same before with my 3.3 RLink and not now because, the new one, was activated into another vehicle. The solution could be to write my vehicle ID into the new RLink but I can't do it, no way to write. I've tryed also with the new function "manual request" but without success, every time with an error.
Do you have an idea for write this value? I need a software modification or it's impossible?
This is my last way for solve the problem
Thanks a lot for all

[image]

[image]

[image]

[image]

[image]

Rcarnero73, 2020-02-26:
I have the same problem.
the cause: the Rlink has the anti theft active and during the boot, check if the VIN in the BCM is = to the VIN in the RLINK
Someone has commented that it could be unlocked by bridging some connection pins, but not sure and not confirmed.

marrius30, 2020-03-11:
any solutions?

StrangerXT, 2020-04-03:
I wish someone could come up with a solution. I'm planning to purchase a used R-Link with an upgraded hw/sw configuration but I think it is impossible to use it in my car since reprogramming VIN / Vehicle ID is impossible.

tiktokup, 2020-07-12:
salut a tous 
ou je peux trouver le vehicle ID du vehicule  dans DDT4 ???
merci

tiktokup, 2020-07-13:
@igno77 

Hi
can you tell me where you are now with your vehicle id problem.
is it resolved or not
thank you so much .

svbfazers, 2020-12-27:
did you finally solve the problem?  I am interested to know since I intend to put a radio from another car in mine

rttr67, 2021-03-05:
Problem solved

Rcarnero73, 2021-03-05:
> Problem solved

How did you solve the problem?
regards

rttr67, 2021-03-06:
hello everything is done on the table intervention inside with the programmer

Tino1973, 2021-03-23:
> Problema resuelto

Hola, como solucionaste el problema. Cambiaste la ID de la radio?

rttr67, 2021-03-23:
i can change radio ID and cancel antiheft

Tino1973, 2021-03-23:
> puedo cambiar la identificación de la radio y cancelar el antihurto 2021-03-23, an, 13:02 Tino1973 ***@***.***> rašė:  
Como, necesito la informacion. HE gastado 400 euros por uno, que ya no puedo úsalo. 
Mi correo es: 
kborisov73@gmail.com

Tino1973, 2021-03-23:
> […](#)
Por favor, dime la solución. Gracias.

tareksadeq, 2021-03-24:
> […](#)

Can you help me too : tito_sadik@hotmail.com 
I need to know how to unlock it

tontito, 2021-04-25:
Can you help : punishcrack@hotmail.com
i want to buy used r-link2

AlMeu, 2021-04-28:
I can help you with antitheft code.

tareksadeq, 2021-04-28:
Yes please , tell me how :)

Thank you for your consideration

sianolo, 2021-05-30:
> Puedo ayudarte con el código antirrobo si es necesario. Te respondo por correo electrónico.

hello, sorry for the inconvenience Could you help me with the antiheft? I'm willing to make up your work please. my email address sianolo.sl@gmail.com

sianolo, 2021-05-30:
Please how do you do the antiheft code sianolo.sl@gmail.com

AndyIreland, 2021-05-30:
Download the 'My Renault' app in the android play store and it should have the radio code in that

rakan10011, 2021-06-09:
> hello everything is done on the table intervention inside with the programmer 2021-03-06, št, 00:05 Rcarnero73 <notifications@github.com> rašė:
> […](#)

I have black screen and tried pyern and nothing who can help me and i pay for it 

email: a556621@gmail.com

rakan10011, 2021-06-09:
> […](#)

I have problem black screen and tried pyern nothing happened my problem trying to downgrade  . Who can help me and I will pay for it 

email: a556621@gmail.com

yaohai68, 2021-06-21:
Can you help  : yaohai68@gmail.com

> 如果需要，我可以帮你防盗码。我通过电子邮件回复您。

Can you help : yaohai68@gmail.com

yaohai68, 2021-06-21:
> 我可以更改无线电 ID 并取消防船 2021-03-23， a， 13：02 蒂诺 1973 *** > rašė：
> [...](#)

Can you help : yaohai68@gmail.com

Pedrom92ps, 2021-07-06:
Its possible help me unlock rlink2? pedrom92@hotmail.com

C0TECH, 2021-07-16:
> 
> […](#)

Can you help me? cotech_i@ukr.net

Fas67, 2021-10-03:
> > 我可以更改无线电 ID 并取消防船 2021-03-23， a， 13：02 蒂诺 1973 *** > rašė：
> > [...](#)
> 
> Can you help : yaohai68@gmail.com
Can you help :
fabio.schili@live.it
Thanks.

artedis1, 2021-10-20:
сan you help please?
artedis@bk.ru

AlvaroDTT, 2021-12-28:
Can you help me pls? I need ti change the ID for remover the anti theft Mode
Alvaro_74_25@hotmail.com

Pekora78, 2022-01-22:
Yo también necesito ayuda con el mismo problema.
daviciyo_78@hotmail.com

rttr67, 2022-01-22:
I can help you with antitheft code if need. I  by email.

Pekora78, 2022-01-22:
Este es mi correo daviciyo_78@hotmail.com

rttr67, 2022-01-22:
All r-link2 blocked i can change the radio id and undo the theft it is done
on the table

rttr67, 2022-01-22:
my test desk

Pedrom92ps, 2022-01-22:
no one will help you, they just sell this service, as there are not many people doing it, they don't reveal how it is.  I know I've seen it done even in the car, they disconnect a plug (I don't know which one) and manage to put the virgin RLINK.  I don't have any car at the moment to do tests but I think it has something to do with disconnecting with the UCH to not check the VIN but I don't know how to do it.

yagokhan, 2022-03-07:
> I can help you with antitheft code if need. I answer you by email.

Can you help me gokhanya@gmail.com

dardoelectro, 2022-03-16:
Hello, I have the problem of the blocked stereo, when I want to change it for the original.
You can help?
I leave you my email trcelectronica@gmail.com

dardoelectro, 2022-03-16:
@rttr67 
Hello, I have the problem of the blocked stereo, when I want to change it for the original.
You can help?
I leave you my email trcelectronica@gmail.com

alquistus, 2022-05-13:
I want to buy a used r-link2 unit and I don't know if there is VIN erased (virgin mode) and if anti-theft is deactivated. Is there someone in Slovakia or near countries who can manage this if there will be a problem? My email: alquistus@gmail.com
Thank you

abdullahhany, 2022-05-20:
hey there please can you help me in unblocking my Rlink?? thank you in advance 
my email 
amona.hany1@gmail.com

AndyIreland, 2022-05-20:
> hey there please can you help me in unblocking my Rlink?? thank you in advance my email [amona.hany1@gmail.com](mailto:amona.hany1@gmail.com)

what is it an r-link 1 or 2 ? - and what car and what is the message on the screen or is the screen just black?

abdullahhany, 2022-05-21:
Hey thank you for replying to me❤️
Its Rlink 2 sw 2.2.19.300 in My kadjar and Rlink 2 sw 9.9 at my friend’s megane iv , we have been stuck for 2 months without radio 
So thank you for helping i really appreciate it❤️

abdullahhany, 2022-05-21:
It just gives me antitheft mode for my car, i changed my Rlink unit with a used one , and he made a wrong update made it black with no response no sound no anything

AndyIreland, 2022-05-21:
oh right sorry, I may have known a fix if it was r-link1 version but not sure about r-link2 , hopefully someone else can answer . - good luck

abdullahhany, 2022-05-21:
Okay can you tell the way for Rlink 1 so i can start from there maybe i can find solution for 2🥲

AndyIreland, 2022-05-21:
no because 2 totally different system - my way involves a file on the SD for the rlink1 system tomtom live

AlMeu, 2022-05-22:
@abdullahhany 
If you have black screen with not sound and USB not working, the unit is dead.
Where are you from? I can help with fresh new units.

abdullahhany, 2022-05-22:
Hi there thanks for your time , yes mu units are dead and i bought a used one but it gives me anitheft on so i wanna solve that can you help?

rttr67, 2022-05-22:
I can help you with antitheft code if need. I by email

svbfazers, 2022-05-22:
Mi radio se quedó muerta a la hora de actualizar el software con la Clip y con el R-ubs, alguien sabría si eso tiene solucion? Se quedo la pantalla en negro, aunque con la Clip si entro en diagnostico con la radio, con DDT4ALL he mirado y el display esta activo pero sigue sin encender

rttr67, 2022-05-22:
Si tiene una pantalla negra sin sonido y el USB no funciona, la unidad está muerta
la unidad de radio necesita ser reemplazada

AlMeu, 2022-05-22:
For those with unit dead, I can help with fresh new units.
Leave an email for contact.

svbfazers, 2022-05-22:
svbfazers@gmail.com

Rcarnero73, 2022-05-22:
Can you help me with the anthitheft code, please?
Thanks in advance

Rcarnero73, 2022-05-22:
> I can help you with antitheft code if need. I by email

Can you Yelp me with the anthitheft code, please?
Ricardo@raiman.net
Thanks in advance

abdullahhany, 2022-05-26:
I need help with antitheft code for Rlink 2 
My email is ***@***.***

loui-19, 2022-05-27:
> I can help you with antitheft code if need. I by email

Hi, I have the same issue and need radio ID changing and anti theft code removing, can you help? powerade2010@hotmail.co.uk  Thanks

DaveFraser, 2022-05-28:
> For those with unit dead, I can help with fresh new units. Leave an email for contact.

Hi AlMeu - I have a dead r-link 2 unit :( please let me know if you can help, thanks!
My email is davilyportser@gmail.com

michalkulesza, 2022-06-25:
@abdullahhany @rttr67  hey, can you guys help me out? kulesza.michaluk@gmail.com

dashcoder13, 2022-07-17:
Hello AlMeu

i need Rlink2 , can you contact me please autokeyprog@gmail.com

thanks

[86 more replies]

closed 2023-06-02 by Furtif