cedricp/ddt4all #260: 2014 Model Dacia dokker Left Turn Signal Lamp Off.

BarektE, 2019-09-21:
How can I turn off and turn on the dacia dokker signal lamps 2014?
Closed from ECU
I want to turn on again with the short signal 
left signal closed ddt4all

[image]

BarektE, 2019-09-22:
those who wrote on this page. they're all empty people. no benefits no.

BarektE, 2019-09-23:
bu sayfanın yararlı olduğunu düşünmüyorum

closed 2019-09-23 by BarektE

canibeyazmehmetali, 2019-11-08:
Merhabalar 
Ben bir konu hakkındaki düşüncelerini öğrenmek istiyorum.
Clio 4 2019 aracımın radio yanlışlıkla sildim ve şuna açılmıyor.
Bu konuda yardımcı olabilir misiniz lütfen

KarelSvo, 2019-12-28:
> How can I turn off and turn on the dacia dokker signal lamps 2014?
> Closed from ECU
> I want to turn on again with the short signal
> left signal closed ddt4all
> 
> [image]

You cannot own a Dokker 2  ;-)