cedricp/ddt4all #262: Dépannage espace 3

mrguen, 2019-09-23:
Bonjour,

Comme beaucoup de personnes j'ai un problème de voyant d'airbag allumé sur mon espace 3 de 1999. Un passage à la valise à montré des défauts en mémoire et le mécanicien ne pouvait pas les effacer mais j'ai l'impression que soit on ne peut pas le faire avec tous les outils soit ils ne savent pas comment. Donc je voudrai tester votre utilitaire

1) Vu que je n'ai pas d'ordi portable je pense utiliser soit wifi soit bluetooth sur mon téléphone Android 5. Vaut-il mieux utiliser Wifi ou bluetooth?

2) Il y a beaucop d'ELM327 dispo sur ebay. Certains sont chinois mais mentionnent le bon MCU exemple https://www.ebay.fr/itm/WIFI-V1-5-PIC18F25K80-chip-ELM327-car-fault-detector-OBD2-z4/113835088795?hash=item1a81199b9b:g:egoAAOSwhzhcBwMP

Est-ce que ça convient? 

3) Je n'ai aucune idée si cela va fonctionner. Mes moyens sont faibles. Je serai heureux de faire un petit don si ça fonctionne. Pouvez-vous débloquer l'application pour essayer?

Cordialement

cedricp, 2019-09-23:
1. Wifi ou Bluetooth, pas de preference de mon cote
2. Il faut etre sur d'acheter avec le bon chip, quitte a le payer un peu plus cher. Sinon il y a le KW902 qui semble marcher systematiquement ou OBDLink.
3. Cela fonctionnera surement, il faudra juste bien interpreter les defauts. Sur Airbag, il s'agit le plus souvent de rupture de ligne (contacteur tournant volant, prises sous les sieges, etc.)
Bien prendre conscience que c'est un outil ingenieur, il ne faut pas l'utliiser stupidement sans reflechir.

mrguen, 2019-09-23:
Je n'y connais rien mais pas l'intention de faire autre chose que ce que vous montrez dans la vidéo d'effacement de défauts d'airbags.

mrguen, 2019-10-03:
Bonjour. J'ai acheté ça https://www.ebay.fr/itm/Konnwei-KW902-ELM327-Bluetooth-OBD2-Car-Diagnostic-Scanner-Code-Reader-Tool-oD/113835204153?ssPageName=STRK%3AMEBIDX%3AIT&var=414061560973&_trksid=p2057872.m2749.l2649 

Mais impossible d'établir une connexion avec l'ECU. J'ai essayé sur plusieurs outils. Votre application s'est lancée, impossible de connecter et maintenant elle crash au démarrage...

docbenton, 2020-01-02:
j'arrive a utiliser ddtall avec cet elm 327 https://fr.aliexpress.com/item/32823514956.html      j'ai eu des pb avec les drivers sous windows 10 et cette page m'a aide http://www.totalcardiagnostics.com/support/Knowledgebase/Article/View/1/0/how-to-install-elm327-usb-cable-on-windows-and-obd2-software

mrguen, 2020-01-02:
Merci pour l'info. En fait l'espace 3 de 1999 ne semble pas compatible. Impossible de communiquer avec l'ECU quelque soit l'application utilisée. Il faut la clip Renault.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif