cedricp/ddt4all #263: Renault Clio MK3 steering wheel angle sensor

surgi, 2019-09-25:
Hello there. 

I'm trying to reset a angle sensor, connected to OBD2 via 1.5 konnei clone, it says I need get status 1 instead of status 4 for getting reserved the sensor by moving steering wheel from left to right ends. Can't get status 1, not appear.

Can you give me some good advice how to sort it out please?

Thanks

BarektE, 2019-09-25:
Fluence ve Megan 3 için bu.

[image]

surgi, 2019-09-25:
Hi.

Thanks for email.

Is that mean I need just press calibration button and turn a steering wheel
to the left untill it stops and then turn right....

No need to press any buttons on OBD2 system?

Thanks

surgi, 2019-09-25:
I have Renault Clio MK3 bro1/cro1 krof fv1krcf0h...... 1.5 DCI 2009.
Recently my car's wheels alignment was done, but steering wheel still need
resetting :(

shrlnm, 2019-09-25:
@surgi Calibration procedure may depend on ECU installed on you car and on type of ABS. Does it equipped with ESP or not. I suggest to find a documentation in dialogys.  

Also you may try pyren. For example below is a snippet from their screen

Prerequisite : OPERATIONS TO BE PERFORMED PRIOR TO CONFIGURATION:
- With the ignition switched on or engine running (to provide maximum steering assistance), turn the steering wheel
  to full left lock and then to full right lock,
- Position the vehicle with the wheels straight, steering wheel at the centre point (straight line driving),
- Run the command by pressing the CONFIRM key.
  NOTE: To obtain accurate straight wheel position (straight ahead driving), this operation must be
  performed on a front axle adjustment bench (particularly for vehicles fitted with ESP).

OPERATIONS TO BE PERFORMED AFTER CONFIGURATION:
- Switch on the ignition and establish dialogue with the electric power 	assisted steering computer (EPAS),
- Turn the steering wheel to full left lock then right lock,
- Verify that the programming takes effect: ET020 Steering wheel angle sensor programming must
  display STATUS 4,
- Check status ET002 Electric PAS immobiliser code. If the status is not present, disconnect the battery to
program the code (see interpretation of status). 
- Verify that the angle value is correct by checking parameter PR121 Steering wheel angle,
- Make sure that there are no faults,
- Perform the necessary repairs by reference to the fault finding manual.
NOTE: if the vehicle is fitted with ESP, establish dialogue with the ABS/ESP computer after the operation; if the function is available (depending on the computer version), delete the ESP sensor programming using command RZ003. Make sure that there are no ESP faults: Perform the necessary repairs by referring to the fault finding manual.

surgi, 2019-09-25:
My model have no esp only abs system.

I have tried pyren, I'm not sure a little bit about shell I adjust a
steering rack length in the system or it just by default deter up.

Thanks

shrlnm, 2019-09-26:
In my opinion, if you do not have ESP, you do not need to calibrate the steering wheel angle, except in order to be in harmony with Feng Shui

surgi, 2019-09-27:
Exactly, I need set it up for Feng Shui purpose only ) I have resetter the
sensor, but it's still 1 or 2 degree out of centre. Font know may be I need
just wait until the ECU will recalibrate it by it self to the right
position.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif