cedricp/ddt4all #264: 2011 Megane 3 Tomtom

BarektE, 2019-09-26:
renault megane tomtom ride. system unconfigured error. How to fix it with ddt4all. because it broke with him.
[image]

BarektE, 2019-09-26:
[image]

cedricp, 2019-09-26:
Would be nice to explain what you did.

BarektE, 2019-09-26:
> Ne yaptığını açıklamak güzel olurdu.

[image]

BarektE, 2019-09-28:
**### **These are the error codes. Is there any help****

DTCStatus.pendingDTC : [00] No
DTCStatus.testFailedSinceLastClear : [01] Yes
DTCDeviceAndFailureTypeOBD : 37697 [0x9341]
DTCFailureType.ManufacturerOrSupplier : 5 [0x5]
DTCFailureType.Category : [05] System Programming Failures
DTCFailureType : [55] not configured
DTCStatus.testNotCompletedSinceLastClear : [00] No
DTCStatus.testNotCompletedThisMonitoringCycle : [00] No
DTCStatusAvailabilityMask : 57 [0x39]
DTCStatus.warningIndicatorRequested : [00] No
DTCStatus.confirmedDTC : [01] Yes
StatusOfDTC : 41 [0x29]
DTCStatus.testFailedThisMonitoringCycle : [00] No
DTCStatus.testFailed : [01] Yes
DTCDeviceIdentifier : [9341] Data

DTC #1
DTCStatus.pendingDTC : [00] No
DTCStatus.testFailedSinceLastClear : [00] No
DTCDeviceAndFailureTypeOBD : 37689 [0x9339]
DTCFailureType.ManufacturerOrSupplier : 3 [0x3]
DTCFailureType.Category : [01] General Electrical Failures
DTCFailureType : [13] circuit open
DTCStatus.testNotCompletedSinceLastClear : [01] Yes
DTCStatus.testNotCompletedThisMonitoringCycle : [00] No
DTCStatusAvailabilityMask : 41 [0x29]
DTCStatus.warningIndicatorRequested : [00] No
DTCStatus.confirmedDTC : [00] No
StatusOfDTC : 16 [0x10]
DTCStatus.testFailedThisMonitoringCycle : [00] No
DTCStatus.testFailed : [00] No
DTCDeviceIdentifier : [9339] CVBS In Line

[image]

shrlnm, 2019-09-28:
I may try to help but I need the dump made in pyren/mod_dd

BarektE, 2019-09-28:
> 

I really need a lot of help. I would be grateful

shrlnm, 2019-09-28:
I am waiting the dump from you

BarektE, 2019-09-29:
> Senden bekliyorum bekliyorum

I did not understand what you're saying

shrlnm, 2019-09-29:
There is other program named pyren
https://gitlab.com/py_ren/pyren
I may try to help you but I need that you get the dump file from r-link module  by pyren program and send it to me.

BarektE, 2019-09-29:
I have a pyren program

BarektE, 2019-09-29:
> Pyren
> https://gitlab.com/py_ren/pyren adında başka bir program var.
> Size yardım etmeye çalışabilirim ama drenaj dosyasını r-link modülünden pyren programı ile alıp bana göndermeniz gerekiyor.

I put a new format on my computer.
All programs will be reinstalled.

Let me reload the Pyren

shrlnm, 2019-09-29:
Make a dump with pyren/mod_ddt and send it me

BarektE, 2019-09-29:
ok

BarektE, 2019-09-29:
> Pyren / mod_ddt ile bir döküm yap ve bana gönder

how do I get the dump from pyren. I do not know that

BarektE, 2019-09-29:
> Make a dump with pyren/mod_ddt and send it me

mine is not r-link but carminat

BarektE, 2019-09-29:
remote connection program. (teamviewer).
I have pyren pc 184 and
android tablet pyren 2.1 board

shrlnm, 2019-09-29:
Ok give me your e-mail

BarektE, 2019-09-29:
> Tamam e-postanı ver

mete.erim@hotmail.com

Burkatronic, 2019-12-01:
Sorry to jump in on this!
Did the problem get resolved?
I've been trying ty fix it for months :/

closed 2022-04-10 by BarektE