cedricp/ddt4all #265: ECU Tweaker: usb connection

pakoto, 2019-09-26:
Hi. How usb connection works? I think that i need OTG female usb adapter, but nothing happens.

cedricp, 2019-09-26:
Did you select device and allow permissions?

pakoto, 2019-09-28:
[image]

Tested using android 7.1.2

Shows "indexing database" half second and close

gobo-ws, 2019-10-08:
I also have problems with USB.
Switching settings from bluetooth to USB in the app, version 1.81 makes it crash. Oneplus 3T running Android 9.
OBDLink SX works fine with Torque on the same phone.

titou4307, 2019-10-21:
Hello at all
I have the same problem with version1.85 and Obdlink SX usb conencted under usb OTG...
App crash

I have activate localisation but bug stay

cedricp, 2019-10-21:
Obdlink is not supported yet, will check that ASAP

titou4307, 2019-10-21:
Sorry Cédric 
My english is poor and I have post in english for all the subject...

so were are frenchy you and me so could you explain in french ? Merci !! Thanks

cedricp, 2019-10-21:
Oui bien sur, il n'y a pas de support pour l'Obdlink encore, il faut que je fasse quelques modifications pour le faire fonctionner, je regarde ca bientot

titou4307, 2019-10-21:
Bonjour Cedric 

Merci d'avance pour le temps passé... 
J'ai vu que la licence était soumise à activation cependant je ne vois pas le "cadenas" 
Peut être apparait il une fois connecté à une interface ? 

J'ai installé l'appli sur le smartphone S9 sous android 9 mais j'ai une vieille tablette sous kit kat (cela marcherait ???) 

Enfin j'ai accès à plusieurs renault (clio 2 perso clio 3 et 4 boulot + d'autres génération) 
=> , si tu veux que je te valides des tests, hésites pas 

Merci encore 

Claude 
----- Mail original -----

cedricp, 2019-10-21:
J'ai désactivé la licence récemment, des version debloquees circulaient deja... Sur Kitkat ca doit marcher, mais uniquement avec de l'ELM, pas eu le temps de travailler sur les autres adaptateurs encore...

GhaithKhmili, 2021-12-10:
Hello @cedricp ,
Any news on the USB crash and OBDLink SX support?
Regards,

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif