cedricp/ddt4all #266: PyRen for windows, guide?

AndyIreland, 2019-09-28:
Hello - there is a setting I want to change in PyRen for windows in the HFM area but its asking for an input 

Press ENTER to exit or type [HEX, DEC, ASCII, VIN]:

and I am stuck I dont know what to do next , I dont want to just press enter because it comes out of it , so what do I choose and then input at this stage? - is there any guide around please? 
[image]

shrlnm, 2019-09-29:
It is a bit complex. Actually the command CF031 have to be run thru configuration scenario "SC007 [Command] REMOTE LOCKING BUZZER" but you run it directly thru cmd-mode.
pyren do not support most scenario execution but you may try to look on this scenario and guess how it works.
For doing this you need to rename the file pyren/scen_ecri_generique2.py into something else.
Then go to the screen SC : Configuration scenarios -> KEYLESS VEHICLE and then choose SC007

`ECU : 11360  FG9911360.XML
Screen : SC : Configuration scenarios -> KEYLESS VEHICLE -> S007 [Command] 

#############################  Command parameters  #############################

Prerequisite : 

name         : S007
codeMR       : SC007
label        : REMOTE LOCKING BUZZER
type         : SC
scenario     : scm:SCEN_ECRI_GENERIQUE2#SCEN_ECRI_GENERIQUE2_11360.xml
inputlist    +
serviceID    +

There is scenarium. I do not support them!!!

Press ENTER to exit or type [SHOW]: show`

Type the word "show" here
Pyren will show you the content of scenarium. 

`
<ScmRoot>

    
    <ScmParam name = "Title" value = "8917"/>

    <ScmParam name = "Tempo_T1" value = "5"/>

    <ScmParam name = "Message1" value = "59096"/>
    <ScmParam name = "Message2" value = "27604"/>
    <ScmParam name = "Message3" value = "13339"/>

    <ScmSet name = "Screens">
      <ScmSet name = "Screenx.1">
         <ScmSet name = "ComboBox1">
           <ScmParam name = "cmdRef" value = ""/>
           <ScmParam name = "defaultStateRef" value = "LC003"/>
           <ScmParam name = "help" value = ""/>
           <ScmParam name = "labelCodeText" value = "8917"/>
           <ScmParam name = "msk" value = "1,10000000"/>
             <ScmSet name = "items">
                <ScmParam name = "16803" value = "10000000"/>
                <ScmParam name = "16802" value = "00000000"/>
             </ScmSet>
         </ScmSet>
      </ScmSet>
     </ScmSet>

    <ScmSet name="Commands">
        <ScmParam name="CF031" value="ID018,ComboBox1.1"/>
    </ScmSet>

` 
There are a lot of parameters like this 'value = "59096"'  for reading it you may use mod_optfile.py tool

`./mod_optfile.py ../Location/DiagOnCan_GB.bqm 59096`

It is a general method how to deal with such commands. Unfortunately pyren can not do everything automatically. But it may give you a hint what and how you may change in ddt4all. 

You may open an appropriate xml file T4_VS_HFM_DDT2000_SW1209_20141126T160247.xml in ddt4all and define parameters there

`      <Request Name="Write_A_AC_Configuration_Data">
        <ManuelSend/>
        <Sent>
          <SentBytes>3B71800000000000000001</SentBytes>
          <DataItem Name="CF_VSC_WAC_Buzzer_inhibited" FirstByte="3"/>
          <DataItem Name="VSC_Warning_Battery_Level_Identifier" FirstByte="5"/>
          <DataItem Name="CF_VSC_CAR_BODY_AND_THATCHAM" FirstByte="9"/>
          <DataItem Name="DONGLE_CF" FirstByte="11"/>
        </Sent>
        <Received MinBytes="2">
          <ReplyBytes>7B71</ReplyBytes>
        </Received>
      </Request>
`

AndyIreland, 2019-10-01:
thank you for the reply shrlnm  - but thats got me right confused , I only know the basics of 'with' or 'without' it PyREN  so it dont looki like I will be changing that then. But thanks.
What does 'WAC'  and 'VSC' mean by the way please?

shrlnm, 2019-10-02:
Unfortunately I do not know what these abbreviations mean

cedricp, 2019-10-02:
VSC means "vehicule sans clef" -> keyless vehicle
Here's a spreadsheet a user sent me with abbrev. mapping :
https://drive.google.com/open?id=0B2LgdbfJUsUZeG9qRHc1aVYyWkFEMFp5NnJpQ3FkRktPNnBN

AndyIreland, 2019-10-02:
> VSC means "vehicule sans clef" -> keyless vehicle
> Here's a spreadsheet a user sent me with abbrev. mapping :
> https://drive.google.com/open?id=0B2LgdbfJUsUZeG9qRHc1aVYyWkFEMFp5NnJpQ3FkRktPNnBN

thats brilliant information thank you cedricp

KarelSvo, 2019-10-03:
> What does 'WAC' and 'VSC' mean by the way please?
WAC is walk away configuration
 Keyless automatically lock and confirm with bugle.
FEEDBACK WAC 1 TM CF and FEEDBACK WAC 2 TM CF  is the length of the bugle sounds, it adjusts the volume.
WAC MUTE TM CF is the pause between the two beeps.
With WAC JINGLE CF you can deactivate the confirmation with the bugle completely.

AndyIreland, 2019-10-03:
> WAC is walk away configuration
> Keyless automatically lock and confirm with bugle.
> FEEDBACK WAC 1 TM CF and FEEDBACK WAC 2 TM CF is the length of the bugle sounds, it adjusts the volume.
> WAC MUTE TM [...]

Thank you, so if i want to disable the 'beep beep' my car makes as I walk away , I should go into T4 VS HFM DDT20000 > VSC Configuration> CF_VSC_WAC _Buzzer_Inhibited and write 'Sans Buzzer' then that should disable the beep beep as I walk away from the car as it locks?

AndyIreland, 2019-10-05:
and is tuning LF1 settings mean different tune for the jingle? 
[image]

AndyIreland, 2019-10-14:
I tried to have a look at 'speed limiter in EMS3155 today in PyREN but when I typed 'show' it just exited back to the menu as if I had just pressed enter. 

[image]

shrlnm, 2019-10-15:
It is probably because you have no EcuReanaul/Scenarios folder

AndyIreland, 2019-10-15:
> It is probably because you have no EcuReanaul/Scenarios folder

ah right i see - how do I get the  EcuReanault/Scenarios folder from please?

shrlnm, 2019-10-15:
pyren has a tool for extract data from installer
https://gitlab.com/py_ren/pyren/blob/master/extract.py

AndyIreland, 2019-10-15:
> pyren has a tool for extract data from installer
> https://gitlab.com/py_ren/pyren/blob/master/extract.py

ah right , Thank you for all your help but i haven't got a clue how to use that tool , so i suppose i am stuck - but thanks again anyway

shrlnm, 2019-10-16:
Find any CLIP distrib in the internet. They are usually in ISO format. On the ISO disk image you will find two files data2.cab and data1.hdr.  No need to install the CLIP. This tool just unpack all necessary files from it.

AndyIreland, 2019-10-16:
OK I have my 183_0_6_0.iso and have extracted the 2 files  data2.cab and data1.hdr and made a new folder called 'for PYREN' and put them into that folder .. now I dont know what to do next thanks

shrlnm, 2019-10-16:
put together 4 files in one folder:

data2.cab
data1.hdr
i12comp.exe 
extract.py

Then simply run extract.py

AndyIreland, 2019-10-16:
so now when I put in 'show' then I just get all this code and still 'press enter to exit'  - stil nothing can be changed then , or is this because I am doing this on Pyren demo and not actually hooked up to car?
[image]
[image]
[image]

KarelSvo, 2019-10-16:
https://transfiles.ru/55r34

EMS 3155 supplemented with cruise screen.
Load this file in ddt4all with the function File/Open xml and check if the read cruise parameters are plausible.

AndyIreland, 2019-10-16:
thank you for the link KarelSvo I will give that a try

BarektE, 2019-10-21:
> WAC

(WAC)
OPENING ELEMENT WARNING SOUND UNIT

AlMeu, 2020-05-31:
I have seen the last PyRen 1.93 for Windos works and interacts with CLIP and DDT2000 database.
[https://gitlab.com/py_ren/pyren](url)
https://www.youtube.com/watch?v=DazsvlnLCoA
I tried but I can't configure to read these in any mode. What I mistake?
[image]
This should be the config for Windows.

AndyIreland, 2020-11-16:
> It is a bit complex. Actually the command CF031 have to be run thru configuration scenario "SC007 [Command] REMOTE LOCKING BUZZER" but you run it directly thru cmd-mode.
> pyren do not support most scenario execution but you may try to look on this scenario and guess how it works.
> For doing this you need to rename the file pyren/scen_ecri_generique2.py into something else.
> Then go to the screen SC : Configuration scenarios -> KEYLESS VEHICLE and then choose SC007
> 
> `ECU : 11360 FG9911360.XML
> Screen : SC : Configuration scenarios -> KEYLESS VEHICLE -> S007 [Command]
> 
> ############################# Command parameters #############################
> 
> Prerequisite :
> 
> name : S007
> codeMR : SC007
> label : REMOTE LOCKING BUZZER
> type : SC
> scenario : scm:SCEN_ECRI_GENERIQUE2#SCEN_ECRI_GENERIQUE2_11360.xml
> inputlist +
> serviceID +
> 
> There is scenarium. I do not support them!!!
> 
> Press ENTER to exit or type [SHOW]: show`
> 
> Type the word "show" here
> Pyren will show you the content of scenarium.
> 
> `
> ```
> 
> <ScmParam name = "Title" value = "8917"/>
> 
> <ScmParam name = "Tempo_T1" value = "5"/>
> 
> <ScmParam name = "Message1" value = "59096"/>
> <ScmParam name = "Message2" value = "27604"/>
> <ScmParam name = "Message3" value = "13339"/>
> 
> <ScmSet name = "Screens">
>   <ScmSet name = "Screenx.1">
>      <ScmSet name = "ComboBox1">
>        <ScmParam name = "cmdRef" value = ""/>
>        <ScmParam name = "defaultStateRef" value = "LC003"/>
>        <ScmParam name = "help" value = ""/>
>        <ScmParam name = "labelCodeText" value = "8917"/>
>        <ScmParam name = "msk" value = "1,10000000"/>
>          <ScmSet name = "items">
>             <ScmParam name = "16803" value = "10000000"/>
>             <ScmParam name = "16802" value = "00000000"/>
>          </ScmSet>
>      </ScmSet>
>   </ScmSet>
>  </ScmSet>
> 
> <ScmSet name="Commands">
>     <ScmParam name="CF031" value="ID018,ComboBox1.1"/>
> </ScmSet>
> ```
> 
> `
> There are a lot of parameters like this 'value = "59096"' for reading it you may use mod_optfile.py tool
> 
> `./mod_optfile.py ../Location/DiagOnCan_GB.bqm 59096`
> 
> It is a general method how to deal with such commands. Unfortunately pyren can not do everything automatically. But it may give you a hint what and how you may change in ddt4all.
> 
> You may open an appropriate xml file T4_VS_HFM_DDT2000_SW1209_20141126T160247.xml in ddt4all and define parameters there
> 
> `
> Request Name="Write_A_AC_Configuration_Data"
> Sent
> SentBytes 3B71800000000000000001
> DataItem Name="CF_VSC_WAC_Buzzer_inhibited" FirstByte="3"
> DataItem Name="VSC_Warning_Battery_Level_Identifier" FirstByte="5"
> DataItem Name="CF_VSC_CAR_BODY_AND_THATCHAM" FirstByte="9"
> DataItem Name="DONGLE_CF" FirstByte="11"
> 
> `

Hello @shrlnm 

back again. I have this page up 

[image]

And I am typing     show    but then it exits and thats it. 

any ideas please?

KarelSvo, 2020-11-17:
Check whether the selected file is available in your Ecu database.

shrlnm, 2020-11-17:
Hello @AndyIreland 
Yes, @KarelSvo  is right. 
Please check that your database is full. 
Check existence of  file EcuRenault/Scenarios/SCEN_LECT_TRACEABILITY1_11258.xml

Gambakg, 2021-02-24:
Hello, Is it possible to unlock Radio unit from Megane 4? Radio has been dismantling from Megane 4 and installed in another car.  Thanks

shrlnm, 2022-11-08:
@lucassssmi The video isn't mine. The function of generation default dump is not in public version yet. I need your VIN and type of ECU.  I'll make the default dump for you.

lucassssmi, 2022-11-08:
@shrlnm  You are lovely man! 
I write you in P.m. **Thanks**

lucassssmi, 2022-11-08:
@shrlnm  there isn't possibility to write you by P.m. by github ... ?
PLease, write me by email geniusmilano@gmail.com 
Thanks.

lucassssmi, 2022-11-10:
@shrlnm  i just solved everything through a guy from the facebook group, he connected remotely and I had the obd inserted. he fixed all the mess I made. Thank you very much for your availability.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

eldimoni1, 2023-10-09:
Hola sé que el tema está un poco abandonado, resulta que active varias cosas con el Dd4all en el Easylink, módulo de la radio , haciendo un puente para evitar el sgw de Renault, es lo único que no puedo activar la cámara trasera, ni con el Dd4all ni con el Pyren , entro en camera sistem config, en los 2 programas,cambio los parámetros, doy activar, reiniciar, y automáticamente uno me dice no data, Dd4all y en el Pyren me dice none, y no se lo que puedo estar haciendo mal , haber si alguien me puede dar un poco de luz, gracias y un saludo

Alex-mtn, 2024-11-30:
> Hola sé que el tema está un poco abandonado, resulta que active varias cosas con el Dd4all en el Easylink, módulo de la radio , haciendo un puente para evitar el sgw de Renault, es lo único que no [...]

Hi, did you find a solution?