cedricp/ddt4all #267: Python2.x is not supported by brew anymore

makayk, 2019-10-01:
The latest version of brew (platform: Mac) is not supporting python2 so  DDT4ALL is not running. The homebrew-core team removed Python2 support. 

Related ticket:
https://github.com/Homebrew/homebrew-core/pull/43602

makayk, 2019-11-11:
Additional information from clean installed MacOS+ Brew

bash-3.2$ python -V
Python 2.7.16
bash-3.2$ python3.7 -V
Python 3.7.4

bash-3.2$ python ./ddt4all-master/ddt4all.py 
Traceback (most recent call last):
  File "./ddt4all-master/ddt4all.py", line 24, in <module>
    import PyQt4.QtGui as gui
ImportError: No module named sip

bash-3.2$ python3.7 ./ddt4all-master/ddt4all.py 
  File "./ddt4all-master/ddt4all.py", line 258
    print name, group
             ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(name, group)?
bash-3.2$

kostyukdv, 2020-01-22:
Any chance to run ddt4all by python3?

makayk, 2021-04-22:
Here is the solution: https://github.com/cedricp/ddt4all/issues/320#issuecomment-680182687

closed 2021-04-22 by makayk