cedricp/ddt4all #268: DDT4ALL / Python 3 branch

makayk, 2019-10-01:
On Mac/brew if I start the app with python3.7 then I receive the following error message:

bash-3.2$ python3.7 ./ddt4all.py 
Traceback (most recent call last):
  File "./ddt4all.py", line 35, in <module>
    import parameters, ecu
  File "/====REMOVEDPATH=====/parameters.py", line 1080
    for key, value in reversed(sorted(labeldict.iteritems(), key=lambda (k,v): (v,k))):
                                                                        ^
SyntaxError: invalid syntax

cedricp, 2019-10-01:
I pushed fixes for that issue, and also graphics loading BTW

Virallata, 2020-01-23:
Now i got this:

  File "ddt4all.py", line 258
    print name, group
             ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(name, group)?

fritzmerwyn, 2020-03-11:
> Now i got this:
> 
> File "ddt4all.py", line 258
> print name, group
> ^
> SyntaxError: Missing parentheses in call to 'print'. Did you mean print(name, group)?

Same for me. I tried to execute ddt4all.py with python and get the same error. I could change the source code to print(...) but that takes like forever. Any help?

makayk, 2021-04-22:
Here is the solution: https://github.com/cedricp/ddt4all/issues/320#issuecomment-680182687

closed 2021-04-22 by makayk