cedricp/ddt4all #269: cedricp bey Ecu Tweaker error

BarektE, 2019-10-03:
cedricp bey son sürümü yükledim Ecu Tweaker durduruldu hatası alıyorum. önceki sürümleri de yüklüyorum yine aynı
[image]

BarektE, 2019-10-04:
who else will know if the author of the program is unknown?

closed 2019-10-04 by BarektE

cedricp, 2019-10-04:
Your ecu.zip must be generated with DDT4All PC version
File menu ->Zip database

BarektE, 2019-10-04:
does not open error and closes. I've already been using

BarektE, 2019-10-04:
This error occurred when you installed the latest version ecu.zip installed

BarektE, 2019-10-04:
telefonu ingilizceye çeviriyorum yine düzelmiyor. daha önce ingilizce yapınca düzeliyordu