cedricp/ddt4all #273: is there any way we can get to change these read only values with DDT4ALL?

AndyIreland, 2019-10-07:
hello - is there any way we can get to change these read only values with DDT4ALL? they seem to be read only , they appear in 'calibration' but in 'configuration' the parameters do not show up allowing to be changed.  Thank you,

[image]

tomidix, 2020-02-08:
bump 
@cedricp I saw that you have mentioned to someone under #99 to restore calibration set of settings (Read_Window_Control_Calibration_Data) which are read only basically. Is there a way to change those settings or was it the configuration screen?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif