cedricp/ddt4all #274: python3 branch: ddt4all with python 3.7

the-guitarman, 2019-10-16:
Hi, I tried to run ddt4all from the python3 branch on my mac with python 3.7.4_1 installed via homebrew. I did not find a workaround to import the module importlib.util:

```
Traceback (most recent call last):
  File "ddt4all.py", line 35, in <module>
    import parameters, ecu
  File "/Volumes/Nifty/ddt4all/parameters.py", line 4, in <module>
    import ecu, elm
  File "/Volumes/Nifty/ddt4all/ecu.py", line 5, in <module>
    import zipfile
  File "/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/zipfile.py", line 8, in <module>
    import importlib.util
ModuleNotFoundError: No module named 'importlib.util'
```

cedricp, 2019-10-21:
Can you try with Python 3.6 ? It's running fine on my Linux box

rbita, 2020-02-20:
Just delete the folder importlib from the project, this is for compatibility with python 2

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif