cedricp/ddt4all #276: Valeo V42 Injection when battery flat

neutrondev03, 2019-10-18:
Hello as some of you might know there is a problem with V4x family injection on new Logan 2. When battery gets flat sometimes the ECU will become virgin and lose sync with UCH.

The only known way so far was to use the 20 digit pin generated on renault.net.

Starting with version V182 of RENOLINK by Gabi he supports this feature.

Anyone has a Renolink? Maybe we can make a plugin for this problem.

I am not really sure how Renolink does it. Is there any other trick to sync the UCH or Gabi had access to the algorithm that generates the 20pin code from the 40pin and VIN? 

Any help is greatly appreciated.

rypkeyscan, 2022-06-02:
Hola, lo pudiste solicionar? tengo el mismo problema

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif