cedricp/ddt4all #277: all of a sudden getting 'no data' in the lighting Edison ECU

AndyIreland, 2019-10-19:
hello my edison ECU was EMM_Edison_DDT2000_SW15.2 V1.0 - now when I do a search with the magnifying glass it wont even find EMM_Edison_DDT2000_SW15.2 V1.0 now and if I manually select Clio IV  and EMM_Edison_DDT2000_SW15.2 V1.0 everything on the lighting down the list in the read boxes are read and have the word NO DATA. 
(I am doing this all in parking mode by the way and my OBD2 Adaptor is picking up all other ECU's fine)
Does it sound like I need the ECU re-programmed at Renault or is there anything else I can do / try? 
All the headlights and fog lights and everything else are all working perfectly OK

I am just trying to think what could of caused this. and all I can think of is the other day i was using PYREN on the laptop and I think I might of gone into lighting on the UPC and done some tests on number 5 'executing tests' 

do you think maybe a disconnection of the battery might get the EMM_Edison_DDT2000_SW15.2 V1.0  showing up again and able to change parameters again? 
Where is the ECU for the EMM_Edison_DDT2000_SW15.2 V1.0  on the Clio IV ? - is it in the engine compartment (that metal one) or is it inside the car next to the heater housing above the cabin filter or the one where the interior fuse box is on a RHD vehicle please? 
Hope you or someone can help on here please I want to be able to get back and change the parameters back on EMM_Edison_DDT2000_SW15.2 V1.0 should I need to some time. 
Thank you. 

in French:
 Learn to pronounce
bonjour mon calculateur edison était EMM_Edison_DDT2000_SW15.2 V1.0 - maintenant, lorsque je lance une recherche à la loupe, il ne trouve même pas EMM_Edison_DDT2000_SW15.2 V1.0 maintenant et si je sélectionne manuellement Clio IV et EMM_Edison_DDT2000_SW15.2 l’éclairage de la liste dans les cases de lecture sont lus et portent le mot NO DATA.
(Je fais tout cela en mode parking, soit dit en passant, et mon adaptateur OBD2 capte bien tous les autres calculateurs)
Est-ce qu'il me semble que l'ECU doit être reprogrammé chez Renault ou puis-je faire / essayer autre chose?
Tous les phares et antibrouillards et tout le reste fonctionnent parfaitement OK

J'essaie juste de penser à ce qui pourrait causer cela. et tout ce que je peux penser, c’est que l’autre jour, j’utilisais PYREN sur un ordinateur portable et j’imaginais peut-être que j’allais éclairer l’UPC et que je faisais quelques tests sur le numéro 5, «exécuter des tests»

pensez-vous que peut-être une déconnexion de la batterie pourrait faire en sorte que l'EMM_Edison_DDT2000_SW15.2 V1.0 réapparaisse et puisse à nouveau modifier les paramètres?
Où se trouve l'ECU pour l'EMM_Edison_DDT2000_SW15.2 V1.0 sur la Clio IV? - se trouve-t-il dans le compartiment moteur (celui en métal) ou à l'intérieur du véhicule, à côté du boîtier de chauffage, au-dessus du filtre de la cabine ou dans celui où la boîte à fusibles intérieure se trouve sur un véhicule à conduite à droite, s'il vous plaît?
J'espère que vous ou quelqu'un peut aider ici s'il vous plaît, je veux être capable de revenir et de modifier les paramètres sur EMM_Edison_DDT2000_SW15.2 V1.0 si j'ai besoin d'un peu de temps.
Je vous remercie.

AndyIreland, 2019-10-20:
sorted - I had the car battery disconnected for around 10 minutes and its back to recognising the ECU now and parameters

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif