cedricp/ddt4all #278: How do you reset / undo any changes?

pwalker1992, 2019-10-20:
How do you reset / undo any changes?
I will pay £50 paypal if you can walk me through it?
I have made changes in DDT4ALL on my Dacia Sandero Stepway and I want to change them back to factory settings.
Thank you!
Philip

cedricp, 2019-10-20:
It may be possible to recover if you share your log files. Look at the file
screen.log, you it's a human readable format file.

pwalker1992, 2019-10-20:
Thank you for your reply.
I have since reinstalled ddt4all so have lost the log files. Is there any way of resetting the ecu? Maybe by restoring from a working file?

AndyIreland, 2019-10-20:
I havent got a Dacia. but i would imagine if some kind soul had the same ecu and version in theirs as yours but untouched they could screen grab their settings and email it to you and then you could put it back to how it was?

pwalker1992, 2019-10-20:
That would be great if someone would do that.
I noticed that in the plugins section, there was a "megane uch reset". Do you know if there is one for dacia? Would this work?

AndyIreland, 2019-10-21:
As cedric said, you dont want to do that because it would erase all settings

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif