cedricp/ddt4all #284: Valeo V40 Mégane/Scenic 3 ECU Virginize

Icereturns, 2019-10-23:
Is DDT4ALL capable of this function. I can see that it can do UCH but can't find ECM.

And before questions are asked, yes I'm sure I want to do this. It's actually a used ECM from another car, I already have my ISK and pin code, I'm already happy with virginising my UCH and ESCL and pairing it to my key card (ESCL was automatic after pairing my UCH and pressing start) I just can't figure Virginising the ECM/ECU.

Alternatively does anyone know how to enable cruise control /speed limiter (CCSL) in a Valeo V40. On the configuration page of my other working ECU that doesn't require Virginising it just says Cruise control [without]

Thanks for your help in advance.

Icereturns, 2019-10-25:
For future reference I was able to virginize the Valeo V40 (1.6 16v VVT) ECU by filling the 95640 eeprom contained internally with FF, until address 0x17CF. Essential I erased most of the data, only leaving two small footer sections. I used a USB CH341A programmer. It is not meant to be able to programme this kind of chip but it worked ok for me.

There are people on the net saying you need to fit the chip backwards when using a CH341A programmer, but I looked at the datasheet for the 95640 and the 25640 (a chip that the CH341A can programme) and found the pin out to be exactly the same (at least for the SOP8 package I was working with).

I used my clip on programming adapter, I connected it as normal, namely pin 1 of the clip to pin 1 of the chip (as per data sheet) I did not modify the programmer from 3.3V to 5V, as suggested elsewhere on the internet because although the chip can take up to 5.5V the datasheet indicates it can operate as low as 2.5V.

I connected the other end of the clip-on to the programmer in the correct and normal way to the 25xxx section of the zif socket.

On the software side I used CH341A programmer version 1.29 and set the chip 25640 5V.

I had no problem with read, but a couple of issues with write most notably modifying individual lines of hex data.

To make it work I

1) read and saved the chip to a bin file
2) erased the chip
3) reopened the saved file and filled the file with "FF" up to address 0x17CF.
4) I wrote the entire "modified" file back to the chip

BarektE, 2019-10-27:
> DDT4ALL bu işlevi yerine getirebilir. Bunun UCH yapabileceğini ancak ECM'yi bulamadığını görebiliyorum.
> 
> Ve sorular sorulmadan önce, evet, bunu yapmak istediğimden eminim. Aslında başka bir arabadan kullanılmış bir ECM, zaten ISK ve pin kodum var, zaten UCH ve ESCL'imi bakire ve anahtar kartımla eşleştirmekten mutluyum (ESCL, UCH'imi eşleştirdikten ve başlat düğmesine bastıktan sonra otomatikti) ECM / ECU’yu Bakire olarak tanımlayamaz.
> 
> Alternatif olarak, Valeo V40'ta seyir kontrolünün / hız sınırlayıcısının (CCSL) nasıl etkinleştirileceğini bilen var mı? Diğer çalışan ECU'm Bakire Olmasını gerektirmeyen konfigürasyon sayfasında, sadece Cruise control [olmadan] yazıyor
> 
> Yardımınız için şimdiden teşekkür ederiz.

Özelden yazın. mete.erim@hotmail.com

Icereturns, 2019-10-27:
Thanks for the offer, but as posted on my second post. I have successfully Virginised the Valeo V40 using a CH341A USB programmer.

Icereturns, 2019-11-04:
For future reference, after Virginising an ECU as you go through the pairing process, just before you press the start button you to enable both cruise control and speed limiter. Turning these on in turn teaches the Virgin ECU it's meant to have these features, the moment you press start the feature set is locked and can not be adjusted afterwards.

konradvanderlinks, 2021-09-26:
Hello!
The car can directly start after virgine like this or must pairing something more like immo , keys etc.....

Icereturns, 2021-09-26:
As stated in my " for future reference post" it needs to be paired with the BCM after being Virgined this is essentially the same as pairing it with the immobiliser. It is easily done with Renolink. Possibly also with DDT4ALL

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif