cedricp/ddt4all #285: what does CSW Failure display stand for ?

AndyIreland, 2019-10-27:
in the lighting section on a clio IV there is a parameter for Nbx_CSW_failure_display_CF .. what does this do / what does CSW stand for - thanks. 

[image]

Icereturns, 2019-10-27:
I'm not sure, but that live doesn't mean it's failed. I think it means it will wait 1 minute before displaying it has failed.

AndyIreland, 2019-11-14:
if I hover over the box I get this:

[image]

now I wonder what ALS stands for then?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif