cedricp/ddt4all #287: Change of R-Link Screen from 7'' to 8'7'' ... Procedure?

MichaelDK2300, 2019-11-01:
Hi,

Can anyone help me?

I have bougt a 8,7' Screen for my R-Link 2 (Talisman).

What settings do I need to change via Developer Mode and/or DDT4ALL.

If I need to tell you software information or hardware information...just let me know.

Many Thanks for your help,
Have a nice weekend,

Best Regards,
Michael / Denmark

shrlnm, 2019-11-02:
Hi
Just apply one of this macro (HIGH1 or HIGH2) with pyren
https://gitlab.com/py_ren/pyren/tree/master/pyren/macro/rlink2

MichaelDK2300, 2019-11-04:
> Hi
> Just apply one of this macro (HIGH1 or HIGH2) with pyren
> https://gitlab.com/py_ren/pyren/tree/master/pyren/macro/rlink2

So basically swap the two screens in my car, and apply either of those two?

Nothing else I need to do before? Software versions on R-Link 2 it wont work with?etc?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif