cedricp/ddt4all #288: ddt4all dont save parameters in section Audio/RadNav 3.3

Ravel2000, 2019-11-04:
Espace 5 RLink 3.3.16 trying to change guidelines for parking assistant doing by ddt4all in expert mode push button Write and see new parameters in right column but after reset they return to previosly. 
Also there is something wrong with radnav section...
[image]

geffroyma, 2021-03-07:
Hello I have the same issue with DDT4all and an Espace V 2015 in version 9.

I just changed the radio and try to setup the guidelines but after the Key on/off the parameters change and goes back to previous.

Is there any solution to modify those values?
Mathieu

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif