cedricp/ddt4all #290: DUSTER HJD AUTO LIGHTS

lisandru06, 2019-11-10:
Hello,
I have the Dacia Duster from 2018 with AUTO indicator on the handle and the light sensor in the dashboard. From begining the auto it works good but by mistake I deactivated from DDT4ALL.

I modified in the **UPC-EMM> T4 EMM VS v2.3> [UPC-EMM] T4 EMM VS v2.3> EXTERNAL LIGHTNING ALS>EXTERNAL LIGHTNING ALS - CONFIGURATIONS** on NBX_AUTO_LIGHT with deactivated. in the past it was a number 98 if I'm not wrong :( and after this the conection is not working anymore. I couldn't put it back to work automatically the lights. Can you send me please a solution for making again working?

Thanks in advance and You are the best in the cars software!

[image]

BarektE, 2019-11-11:
[image]

lisandru06, 2019-11-13:
I've tested again today and i compare with an original software.. the 2 settings are where i upload the picture but i cannot read and modify like reference due to out of range..
[image]
[image]

lisandru06, 2019-11-13:
[image]

KOPRajs, 2021-08-28:
Hi, I've just uploaded the complete default settings of this screen for my Dacia Duster 2020 here:
https://github.com/cedricp/ddt4all/issues/609

Do you remember which settings have you changed? What were you trying to accomplish?

I'm going to try to adjust the auto light sensor sensitivity so I hope not to break it as well.

KOPRajs, 2021-08-28:
P.S. You might want to look at the following screen, maybe you can set the Nbx_Auto_Light_CF from there:
[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif