cedricp/ddt4all #292: cannot write data on ECU Tweaker

gwansik, 2019-11-13:
First of all, I'd like to thank cedricp for developing this great work.

I am trying to activate Apple Carplay on my Renault Koleos.
I failed to connect my laptop to an OBD2 bluetooth adapter.

But I was able to connect my android phone to the adapter. So I downloaded the android ported ECU Tweaker and found the right configuration page. After I authorized Carplay and AAP, I tried to write data but it didn't work. 

After checking the log file, I found it actually send the bytes for the "write". But after that, an error came up. Hope someone help me please.

New session with ECU RadNav 2.X v2.5 
CAN SENT: [13-11-10:03:10] 1003 
CAN RECV: [13-11-10:03:10] 5003003201F4 
CAN SENT: [13-11-10:03:10] 222203 
CAN RECV: [13-11-10:03:10] 62220300 
CAN SENT: [13-11-10:03:10] 222202 
CAN RECV: [13-11-10:03:10] 62220200 
CAN SENT: [13-11-10:03:10] 222130 
CAN RECV: [13-11-10:03:10] 622130B10F0115293D55 
CAN SENT: [13-11-10:03:15] 1003 
CAN RECV: [13-11-10:03:15] 5003003201F4 
CAN SENT: [13-11-10:03:16] 2E2130B16F0115293D55 
CAN RECV: [13-11-10:03:16] ERROR : 1002E2130B16F01 NO DATA 2115293D55 NO DATA

cedricp, 2019-11-13:
You better try to do it with the PC version, Android version need some more work to support long frames upload (software CFC). There's a try in the latest version, but it's untested yet.

gwansik, 2019-11-14:
Thanks for the reply.
Like I said, I failed to connect my laptop to obd2 bluetooth adpater.
After I selected the COM Port, it cannot open the chosen com port.

Same thing happens with ScanMaster-elm software. But, if I change the interface type Serial to Bluetooth, it can connect.

Think my laptop has some problem with Bluetooth - serial port communication.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif