cedricp/ddt4all #293: DDT4ALL and USBtin

nishimura-san, 2019-11-13:
anybody could tell me how ro remove the ELM327 check or implement USBtin in DDT4all? This thing is a simple USB (registers in windows as a COM port and acts like one) to CAN interface with an API which seems easily adapted for other apps. But me being a noob in programming, would like some pointers on how to implement this interface in DDT4ALL's interface to have an alternative to ELM327 and the infamous Derelek and the CAN CLIP.

I am not requesting a pre-made program, just some pointers. Would also like for @cedricp to point me somewhere. any hint is a good one. but keep in mind i am no programmer. 

Python implementation of the API is here https://github.com/fishpepper/pyUSBtin

cedricp, 2019-11-13:
This device seems based on the MCP2515 chip, and there's no CAN-ISOTP support by default. You'll need to first implement software ISOTP protocol and then the hardware bindings.
It's a quite big task, but possible. Example here : https://github.com/openxc/isotp-c

nishimura-san, 2019-11-14:
and how would someone go about implementing ISO-TP? where exactly in the source code is the communication layer happening?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif