cedricp/ddt4all #295: cannot change parameters of lighting on EMM_EDISON_DDT2000_SW15_2_V1.0 with ECU Tweaker

AndyIreland, 2019-11-13:
hello - for some reason I cannot set parameters for lighting configuration using ECU tweaker  version V1.81   I can change parameters for other things on that ECU like settings for wash & wipe but just not the settings under the 'lighting' section - have tried also changing CAN timeout to 600ms

I can however change the lighting parameters using  DDT4ALL and a laptop - just not in ECU tweaker for android

here is log file:

[logs_[13-11-10_46_49] .txt](https://github.com/cedricp/ddt4all/files/3843873/logs_.13-11-10_46_49.txt)

[image]

oktaybilen, 2021-07-01:
I wonder if anyone has changed the lighting configuration from the old versions of ecu tweaker?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif