cedricp/ddt4all #296: Oil Level

tanerdogan, 2019-11-14:
Hi
Is it possible to see oil level with ddt4all? 
Thanks.

AndyIreland, 2019-11-15:
+1
I used to like seeing oil level in our Clio Mk2 when you started the engine and the reassuring "Oil OK' 
however in DDT4all I enabled 'with oil level' in the cluster settings and still cannot see anything to do with oil on the cluster info window (maybe there will be if the oil level goes low?, maybe a message will pop up ?)

tanerdogan, 2019-11-15:
Same as Clio IV 2015 too. Its just says on the screen when oil level is low... I think 2016 after they can see Renault Go app with MediaNav bluetooth connection but not work before 2016... I believe its read CAN BUS protocol.   Hmmm if i can connect MediaNav BT without ODB and *if possible to* read, send CAN command, maybe I can make a mobile app or WinCE app with that.

PS. Sorry for my English...

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif